steveicarus/iverilog #560: tranif1 not working

jzhang2004, 2021-11-07:
The simple modeling of an sram cell using transif1 switch not working properly... I got this example from Moorby Thomas text book on verilog... the issue is that the transif1 switch can't set the value of  cross coupled inverters (the storage element),  the default drive strength should be strong drive (I think) which should the win the battle again (pull0, pull1) strengthed inverters...

/* the code of an sram cell modeling*/
module sram (output dataOut, 
	input address, dataIn, write);
 tri w1, w3, w4;
 bufif1 g1(w1, dataIn, write);
 tranif1 g2(w4, w1, address);
 not (pull0, pull1) g3(w3,w4),g4(w4,w3);
 buf g5(dataOut,w1);
endmodule

module wave_sram #(parameter d=100);
wire dataOut;
reg addr, dataIn, write;

sram c(dataOut, addr, dataIn, write);

initial
begin
	$dumpfile("sram.vcd");
	$dumpvars;
	#d dis;
	#d addr=1;    #d dis;
	#d dataIn=1;  #d dis;
	#d write=1;   #d dis;
	#d write=0;   #d dis;
	#d write='bx; #d dis;
	#d addr ='bx; #d dis;
	#d addr = 1;  #d dis;
	#d write =0;  #d dis;

end

task dis;
 $display($stime,, "addr=%v d_In=%v write=%v d_out = %v ",
	 addr, dataIn, write, dataOut,
	 "(134)=%b%b%b", c.w1, c.w3, c.w4,
	 "  w134=%v %v %v", c.w1, c.w3, c.w4);
endtask
endmodule

martinwhitaker, 2021-11-07:
The problem is not in the implementation of the tranif, but in the way vvp performs initialisation of the structural network. Initial values are taken from constants and from register variables and propagated through the network. But in this case there is a loop (through w3 and w4) that is not fed by anything and hence is not initialised correctly. And until all the inputs of the tranif are initialised, it cannot resolve and propagate the final values.

The underlying fault can be observed in a simpler example:
````
module test;

wire n1, n2;   

not (pull0, pull1) g1(n2, n1), g2(n1, n2);

initial begin
  #1 $display("%v %v", n1, n2);
end

endmodule
````
which should display `PuX PuX` but instead displays `HiZ HiZ`.

I'm not sure how best to fix this.

jzhang2004, 2021-11-08:
any work around that would put the value of n1 and n2 to x? Trireg is not supported obviously... I found work arounds btw.. either way I will play around and hope to find some work arounds

martinwhitaker, 2021-11-08:
A workaround is to add this to your sram module:
````
 initial begin
    force w4 = 1'bx;
    release w4;
  end
````

jzhang2004, 2021-11-08:
that worked, and I learned bout force and release! thanks!