steveicarus/iverilog #562: Systemverilog array assignment support

TheZoq2, 2021-11-11:
I'm not sure if this is a bug or a feature request, but I have the following code:

```verilog
    logic[63:0] data[6:0];
    initial begin
        data = '{
            64'd0,
            64'd1,
            64'd2,
            64'd3,
            64'd4,
            64'd5,
            6
        };
    end
```
or
```verilog
	logic[63:0] data[6:0] = '{
            64'd0,
            64'd1,
            64'd2,
            64'd3,
            64'd4,
            64'd5,
            6
        };
```
both of which give the following error:
```
arraytest.sv:3: error: Cannot assign to array data. Did you forget a word index?
```

As far as I can tell, this is valid verilog and is accepted by verilator (at least in lint mode, I haven't tested a full compilation). This would be really useful to have implemented in my opinion, and even if that's not a lot of work it might be nice with a more clear error message. Reading the source code for where that error message is defined, it looks like it's talking about something else.

I'd be happy to look into supporting this construct if desired, though I might need some guideance to know where to look

martinwhitaker, 2021-11-11:
The error message was written before the compiler supported SystemVerilog. Traditional Verilog only allows you to assign to individual words in an array, which is what the error message is hinting at. We should output a "sorry" message when compiling SystemVerilog. And the original message would be clearer if it put quote marks around the variable name.

It might not be too hard to support this functionality. We could use the same approach used to implement foreach statements, which the compiler converts into traditional for loops during elaboration.

martinwhitaker, 2021-11-12:
For now I've just improved the error message.

PR #953 (2023-06-19): Add initial support for array assignment patterns
SystemVerilog allows to use assignment patterns to assign values to an
array. E.g. `int a[4] = '{1, 2, 3, 4}`.

Each value is evaluated in the context of the element type of the array.

Nested assignment patterns are supported. E.g. `int a[2][2] = '{'{1, 2},
'{1, 2}};`

Add initial support for array assignment patterns for both continuous as
well as procedural assignments.

For continuous assignments the assignment pattern is synthesized into an
array of nets. Each pin is connected to one of the assignment pattern
values and then the whole net  [... cut]
closed 2023-06-19 by caryr