steveicarus/iverilog #564: Verilog-AMS supported features

GustavoAle, 2021-11-28:
I see in the iverilog man-pages that "_Very little Verilog-AMS specific functionality is currently supported._" using the flag `-gverilog-ams`. At the time I'm unable to compile any simple analog or mixed-signal design. 

Where can I find exactly what can I do with analog signals in icarus? And how can I help to expand the support to it?

GustavoAle, 2021-11-30:
Just to add some details in hope someone answer this ticket. The error I got when trying to compile is:
`res.vams:10: assert: elab_expr.cc:1679: failed assertion 0`
For the following code:
```
`include "disciplines.vams"

module resistor (p, n);
	input real p;
	input real n;
	electrical p, n;
	parameter real r=1.0;

	analog 
		V(p,n) <+ I(p,n)*r;
		
endmodule
```

martinwhitaker, 2021-12-02:
My usual starting point for seeing what's supported (other than just trying it) is to look in the test suite (https://github.com/steveicarus/ivtest). Look in the file regress-vams.list for the Verilog-AMS specific tests. As implied by the manual page, there aren't very many of them. Note that the analog1 and analog2 tests are marked as known failures in the expected results.

Compiling your example using the head of the master branch, it no longer stops at an assertion failure:
````
% iverilog -gverilog-ams test.v
test.v:10: error: No function named `V' found in this context (resistor).
test.v:10: error: No function named `I' found in this context (resistor).
test.v:10: error: The l-value of a contribution statement must be a branch probe access function.
Elaboration failed
````
I'm not familiar with Verilog-AMS, so can't tell if that's an error in your code or something lacking in the compiler.

caryr, 2021-12-03:
The Verilog-A looks fine. As I remember contribution statements in general are not functional and the biggest issue is there is only the underlying digital simulation engine. Things that make sense in that context often work, but most Verilog-A requires support for continuous time simulation to function correctly.

Like Martin said "The test suite is a good indication of what is supported."

martinwhitaker, 2024-01-08:
Closing as answered.

closed 2024-01-08 by martinwhitaker