steveicarus/iverilog #567: SystemVerilog: Generate do not output error

cfcfty, 2021-12-10:
Hi!

I have a code:
```
module arithmetic_right_shift_of_N_by_S_using_for_inside_generate
#(parameter N = 8, S = 3)
(
  input  [N - 1:0] a, 
  output [N - 1:0] res
);

  wire signed [N - 1:0] a_s = a;
  
  genvar i;

  // Bad version
  generate
    for (i = N - 1; i > -1; i --)
      assign res[i] = i > (N - 1 - S) ? a_s[N - 1]
                                      : a_s[i + S] ;
  endgenerate

 //True version

  /*
   generate
      for (i = 0; i < N; i ++)
        if (i >= N - S)
          assign res [i] = a_s[N - 1];
        else
          assign res [i] = a_s[i + S];
    endgenerate
  */
  
endmodule
```
According to the Verilog-2001 standard, all "genvar" variables are unsigned.
They cannot take negative values, but iverilog successfully compiles the module, although it should give an error. 
Also, according to a friend, I know that iverilog can sometimes hang.
Full file with testbench you can see in: https://github.com/cfcfty/verilog_test/blob/main/p017_arithmetic_shift_or_signed_divide_by_power_of_2.v
I use IVerilog 11.0 stable on Linux and setting ` -g2005-sv `.

Thanks.

martinwhitaker, 2021-12-10:
The restriction on negative genvar values was removed in IEEE 1364-2005, and IEEE 1800-2005 includes an example that relies on that (section 19.12.5: Unpacked array ports and arrays of instances). So no error should be reported.

Admittedly the compiler doesn't report an error if you compile this code with the -g2001 option. So I will leave this open as an enhancement request. We would probably make it a warning, not an error, as it has no impact on valid Verilog-2001 code.

We can't do anything about your comment that iverilog sometimes hangs unless you or your friend can provide a simple test case that demonstrates the problem. If you can, please open a new issue for that.

yuri-panchul, 2021-12-11:
> 

Here is a minimal case for the hanging problem.

Processor	AMD Ryzen 5 PRO 4650U with Radeon Graphics   2.10 GHz
Installed RAM	8.00 GB (7.37 GB usable)
System type	64-bit operating system, x64-based processor

Edition	Windows 10 Home
Version	21H1
OS build	19043.1348
Serial number	MP1W8KCH
Experience	Windows Feature Experience Pack 120.2212.3920.0

Icarus files:

Directory of C:\iverilog\bin

02/04/2021  10:33 AM            50,688 iverilog-vpi.exe
02/04/2021  10:33 AM            87,552 iverilog.exe
12/22/2020  11:31 AM            99,146 libbz2-1.dll
12/09/2020  06:50 PM            82,097 libgcc_s_seh-1.dll
01/08/2021  02:31 PM            63,530 libhistory8.dll
01/08/2021  02:31 PM           248,987 libreadline8.dll
12/09/2020  06:50 PM         1,752,711 libstdc++-6.dll
08/04/2020  02:15 AM            45,347 libtermcap-0.dll
12/28/2020  05:20 PM            58,539 libwinpthread-1.dll
02/04/2021  10:33 AM           961,536 vvp.exe
01/05/2021  10:54 AM           116,428 zlib1.dll

Command line:

This hangs:

iverilog -g2005-sv t1.sv

t1.sv:

module m (input [7:0] a, output [7:0] res);
  
  genvar i;

  generate

  for (i = 7; i >= 0; i --)
    assign res [i] = a [i];

  endgenerate

endmodule

This does not hang:

iverilog -g2005-sv t2.sv

t2.sv:

module m (input [7:0] a, output [7:0] res);
  
  genvar i;

  generate

  for (i = 0; i < 8; i ++)
    assign res [i] = a [i];

  endgenerate

endmodule

Stderr:

warning: verinum::as_long() truncated 32 bits to 31, returns 2147483647
^C^C

yuri-panchul, 2021-12-11:
I will add it as a new issue

martinwhitaker, 2021-12-13:
Fixed in both the master and v11 branches.

closed 2021-12-13 by martinwhitaker