steveicarus/iverilog #568: Compilation hangs when a generate loop is used with -- applied to genvar

yuri-panchul, 2021-12-11:
Compilation hangs when a generate loop is used with -- applied to genvar

Processor	AMD Ryzen 5 PRO 4650U with Radeon Graphics   2.10 GHz
Installed RAM	8.00 GB (7.37 GB usable)
System type	64-bit operating system, x64-based processor

Edition	Windows 10 Home
Version	21H1
OS build	19043.1348
Serial number	MP1W8KCH
Experience	Windows Feature Experience Pack 120.2212.3920.0

Icarus files:

Directory of C:\iverilog\bin

02/04/2021  10:33 AM            50,688 iverilog-vpi.exe
02/04/2021  10:33 AM            87,552 iverilog.exe
12/22/2020  11:31 AM            99,146 libbz2-1.dll
12/09/2020  06:50 PM            82,097 libgcc_s_seh-1.dll
01/08/2021  02:31 PM            63,530 libhistory8.dll
01/08/2021  02:31 PM           248,987 libreadline8.dll
12/09/2020  06:50 PM         1,752,711 libstdc++-6.dll
08/04/2020  02:15 AM            45,347 libtermcap-0.dll
12/28/2020  05:20 PM            58,539 libwinpthread-1.dll
02/04/2021  10:33 AM           961,536 vvp.exe
01/05/2021  10:54 AM           116,428 zlib1.dll

Command line:

This hangs:

iverilog -g2005-sv t1.sv

t1.sv:

module m (input [7:0] a, output [7:0] res);
  genvar i;
  generate
  for (i = 7; i >= 0; i --)
    assign res [i] = a [i];
  endgenerate
endmodule

This does not hang:

iverilog -g2005-sv t2.sv

t2.sv:

module m (input [7:0] a, output [7:0] res);
  genvar i;
  generate
  for (i = 0; i < 8; i ++)
    assign res [i] = a [i];
  endgenerate
endmodule

Stderr:

warning: verinum::as_long() truncated 32 bits to 31, returns 2147483647
^C^C

martinwhitaker, 2021-12-11:
Your example compiles without error on my Linux system, but I can reproduce the fault in a Windows 10 VM. It occurs with both 32-bit and 64-bit builds of the compiler.

martinwhitaker, 2021-12-13:
Fixed in both the master and v11 branches.

closed 2021-12-13 by martinwhitaker