steveicarus/iverilog #570: Initialize array via function within package

logicmonkey, 2021-12-20:
```
package mypkg;
  localparam ELEM = 4;
  typedef int myints_t [ELEM-1:0]; // create a type that is an array of ints

  localparam myints_t myarr = init_arr(ELEM); // initialise constant as an array of values set by a function

  function automatic myints_t init_arr(int N); // an easy function
  begin
    for (int i=0; i<N; i++)
      init_arr[i] = i;
  end
  endfunction
endpackage

module dummy;
import mypkg::*;
  initial begin
    for (int i=0; i<mypkg::ELEM; i++)
      $display(mypkg::myarr[i]);
    $display(mypkg::ELEM);
    $finish;
  end
endmodule
```

Expected output (as per commercial simulator):
```
xcelium> run
          0
          1
          2
          3
          4
Simulation complete via $finish(1) at time 0 FS + 0
```

Actual output (some rev 12 hash build run with iverilog -g2012)
```
0
1
0
1
          4

```

martinwhitaker, 2021-12-20:
Icarus Verilog does not currently support unpacked arrays as (local) parameters or function return values, or indeed, most other places where they can't be used in traditional Verilog. But it should warn you about this and fail to compile your code, rather than producing incorrect results.

logicmonkey, 2021-12-21:
I tried enforcing PACKED types - same problem (ATTEMPT1)

```
package mypkg;
  localparam ELEM = 4;

  typedef logic  [$clog2(ELEM)-1:0] elem_t;     // pack this
  typedef elem_t [ELEM-1:0]         mypints_t;  // and this

  localparam mypints_t ATTEMPT1 = init_arr(ELEM);  // initialise constant as an array of values set by a function

  function automatic mypints_t init_arr(int N); // an easy function
  begin
    for (int i=0; i<N; i++)
      init_arr[i] = i;
  end
  endfunction

  //typedef int myints_t [ELEM-1:0];                // unpacked array of ints
  //localparam myints_t ATTEMPT2 = '{3, 2, 1, 0};

  //localparam logic [ELEM-1:0][$clog2(ELEM)-1:0] ATTEMPT3 = 8'b11_10_01_00;

endpackage

module dummy;
import mypkg::*;
  initial begin
    for (int i=0; i<mypkg::ELEM; i++)
      $display(mypkg::ATTEMPT1[i]);

    //for (int i=0; i<mypkg::ELEM; i++)
    //  $display(mypkg::ATTEMPT3[i]);

    $finish;
  end
endmodule
```

ATTEMPT2 and ATTEMPT3 also fail, which I was hoping would at least allow me to write a script to generate the array of int values that I want. Any work arounds gratefully received.

martinwhitaker, 2021-12-23:
Yes, having looked, I now find that multi-dimensional packed arrays are not supported in parameters/localparams.
Do you actually need the array to be a localparam (i.e. do you need to use it in a constant expression)? Using a variable instead, e.g.
````
package mypkg;
  localparam ELEM = 4;

  typedef logic [ELEM-1:0][$clog2(ELEM)-1:0] parray_t ;

  parray_t array = { 2'd3, 2'd2, 2'd1, 2'd0 };
endpackage
````
almost works, although it hits another bug when `array` is imported into a module (which may be easier to fix).