steveicarus/iverilog #571: multiplication won't work

jjts, 2021-12-22:
wire [2*RHO_F_W+1:0] alpha_down_mul;
wire [2*RHO_F_W+1:0] alpha_up_sel_ext = {{RHO_F_W+1{1'b0}}, alpha_up_sel};
wire [2*RHO_F_W+1:0] rho_f_ext = {{RHO_F_W+1{1'b0}}, rho_f};
assign alpha_down_mul = alpha_up_sel_ext * rho_f_ext; //2 Q 2*RHO_F_W

alpha_up_sel_ext is 1<<30, rho_f_ext has data in its 30 lsbs

result is 0! I expected rho_f_ext to be shifted 30 bits to the left

martinwhitaker, 2021-12-22:
I can't tell what's wrong from this incomplete code fragment. What are the declarations for `RHO_F_W`, `rho_f`, and `alpha_up_sel` ?

jjts, 2021-12-22:
Sorry, it is indeed incomplete, but this is not iverilog's problem -- I found the reason later: it is a Gtkwave issue when visualizing large fixed-point numbers.... I am closing this issue and thanks a lot for the fast reaction.

closed 2021-12-22 by jjts