steveicarus/iverilog #575: Using constant value from an interface

RossDMartin, 2022-01-04:
I'm trying to get a constant value out of an interface to use at the next level up in the hierarchy.  The reason for this is I have a module instantiating memory, the memory has different read delay depending on its size, and the next level up needs to know the delay without violating encapsulation.

I understand that constant values are supposed to be extractable from interfaces (?), although this seems more likely to work using $bits().  I have come up with the following simple test case showing a failure, a bug, and a workaround:

```
//
// Want to get the important calculated value A out of the interface.
//
interface subm #( parameter X=0, parameter Y=0 ) ();
   parameter       A = X+Y;
   parameter [A:1] B = 0;
   reg [A:1]       C;
endinterface

module top();

   subm #( .X(1), .Y(1000) ) sub ();

   localparam CT1 = -1; // sub.A;   // sub.A fails here at compile time with Icarus 11.0
   localparam CT2 = $bits(sub.B);
   localparam CT3 = $bits(sub.C);

   initial $display("\n\n\nCompile-time value of A is %0d, %0d, %0d", CT1, CT2, CT3);
   initial $display("Run-time value of A is %0d, %0d, %0d\n\n", sub.A, $bits(sub.B), $bits(sub.C));
   
endmodule // top
```

In Icarus Verilog 11.0 this prints:

```
Compile-time value of A is -1, 1001, 0
Run-time value of A is 1001, 1001, 1001
```
It seems to me that all of these statements should compile and that all of them should give the same results.

That may be arguable, but it's certainly a bug for $bits(sub.C) to return 0 to a localparam -- I believe it should return a correct value, but if not then it should return a compile-time error rather than returning a bad value.

Also, it seems that Icarus has everything in place to allow sub.A to be treated as a constant and extracted from the interface, since $bits(sub.B) effectively does this.  So perhaps it's not too hard to make sub.A work in a future version?

In any case, $bits(sub.B) seems to be a workaround to my problem, although it's ugly.  This may help others with a similar problem.  Hopefully it synthesizes!

martinwhitaker, 2022-01-04:
This looks like issue #483.

martinwhitaker, 2024-02-11:
See discussion #1100.

martinwhitaker, 2024-04-06:
As discussed in #1100, the compiler is correct to reject `localparam CT1 = sub.A` . Currently the compiler does allow hierarchical names as arguments to the `$bits` system function in a constant expression, and I am not minded to change that. The anomalous behaviour of `localparam CT3 = $bits(sub.C)` is now fixed in the master branch.

closed 2024-04-06 by martinwhitaker