steveicarus/iverilog #576: Assertion when calling $clog2

msinger, 2022-01-06:
Calling
```
iverilog -g2012 -o test test.sv
```
with this `test.sv`:
```
module test;
	function void fun;
		$clog2(1);
	endfunction
	always_comb fun;
endmodule
```
fails with an assertion:
```
test.sv:5: warning: always_comb process has no sensitivities.
ivl: net_scope.cc:502: const NetTaskDef* NetScope::task_def() const: Assertion `type_ == TASK' failed.
sh: line 1: 11118 Done                    /usr/local/lib/ivl/ivlpp -L -F"/tmp/ivrlg25dedf82f" -f"/tmp/ivrlg5dedf82f" -p"/tmp/ivrli5dedf82f"
     11119 Aborted                 | /usr/local/lib/ivl/ivl -C"/tmp/ivrlh5dedf82f" -C"/usr/local/lib/ivl/vvp.conf" -- -
```

Replacing `always_comb` with `always` or `initial` works fine. Calling `$clog2` within an `always_comb` block directly (without being wrapped in another function) also works. It prints this warning though:
```
warning: System task ($clog2) cannot be synthesized in an always_comb process.
```
Is this true? Shouldn't this get optimized away during synthesis when the argument to `$clog2` is constant at compile time?

martinwhitaker, 2022-01-23:
The compiler doesn't handle functions called as tasks when calculating the always_comb sensitivity list. That is indeed a bug. The warning when you call $clog2 directly is probably a consequence of that.

However, I struggle to see any practical use for calling $clog2 and discarding the result.

martinwhitaker, 2022-01-23:
See related issue #539.

msinger, 2022-01-25:
Thank you for explaining. I was aware that void functions are treated as tasks and are not included in the sensitivity list, but I wasn't aware that this also is the case for non-void functions when I ignore the return value.
I came up with this test case by stripping down a much much bigger piece of code, that caused the ```Assertion `type_ == TASK' failed.``` error. In that code I actually used the return value. So I think I must search again for what's causing this.

My primary issue is this assertion. My sensitivity lists were ok I think. Is this assertion triggering, because I'm not allowed to call tasks (in this case $clog2 without checking the result) from within a function (fun in my example)?

PR #629 (2022-02-28): Always comb vs void func
This fixes the assertion failure that happens when void functions are used in always_comp functions. This fixes #576, but does not address the problem of the contents of a void function not being scanned for sensitivity lists.
closed 2022-02-28 by steveicarus