steveicarus/iverilog #580: Parsing Verilog to blif

Mbonea-Mjema, 2022-01-11:
I am running into an error

 ivl: main.cc:834: void read_sources_file(const char*): Assertion `0' failed.
Aborted (core dumped)

[image]

Here is the verilog code I want to convert.
<script src="https://gist.github.com/Mbonea-Mjema/033a5eb2c25437e3206782c66bf1b715.js"></script>

martinwhitaker, 2022-01-23:
I don't know if @steveicarus still uses and maintains the blif target. I've never used it myself.

steveicarus, 2022-01-24:
I don't use or actively maintain it. I'm willing to entertain reports about
it, but I'm not going to wade through a huge example program like you
linked to in the report. Attach the smallest possible example, and you're
more likely to get support.

steveicarus, 2022-02-28:
PR#630 is an attempt to replicat this issue, but the sample source file seems to compile just fine with the current master branch (with PR#630) so this issue is not reproducible. RE-open if you feel otherwise.

closed 2022-02-28 by steveicarus