steveicarus/iverilog #608: IVERILOG_VPI_MODULE_PATH is not parsed correctly

DavidC-75, 2022-02-10:
I've been experimenting a bit with IVERILOG_VPI_MODULE_PATH recently, and it didn't seem to work the way I expected.
Looking at the details, I believe there are two issues in `vvp/vpi_modules.cc`.
This is all in the same area:
```
     if (char *var = ::getenv("IVERILOG_VPI_MODULE_PATH")) {
            char *ptr = var;
            char *end = var+strlen(var);
            int len = 0;
            while (ptr <= end) {
                  if (*ptr == 0 || *ptr == ':' || *ptr == ';') {
                        if (len > 0) {
                              vpip_add_module_path(strndup(var, len));
                        }
                        len = 0;
                        var = ptr+1;
                  } else {
                        len++;
                  }
                  ptr++;
            }
      }
```

- the first issue is platform-specific. The colon character ':' cannot be used as a separator on Windows. A simple #ifdef __MINGW32__ can be introduced to exclude it.
- the second issue affects all platforms. The current code seems to assume that all directories listed in IVERILOG_VPI_MODULE_PATH are terminated with a final irrelevant character ( '/' or '\'), which may not always be the case. What seems to work reliably for me is to change the code to `vpip_add_module_path(strndup(var, len+1));`

martinwhitaker, 2022-03-25:
I agree the first issue is a bug. I can't reproduce the second issue when running under Linux. What platforms did you test on?

martinwhitaker, 2022-03-25:
The second issue was a bug in the Windows specific code, so I don't believe it affects other platforms. That bug is now fixed.

DavidC-75, 2022-03-25:
> What platforms did you test on?
That was Windows - I just tried on Linux and it works as expected. 

> The second issue was a bug in the Windows specific code
I didn't realize that strndup was implemented manually for Windows, and that the bug was there.... My quick assessment in the OP was not quite right !

> I agree the first issue is a bug.
This one remains but isn't hard to fix ;-)

DavidC-75, 2022-03-25:
> This one remains but isn't hard to fix ;-)

My mistake - Just realized it's fixed too !
I guess you can close this bug now.

martinwhitaker, 2022-03-25:
The first issue is fixed in the master branch. I won't backport it to v11 because it is a change in behaviour, and it's possible it might break someone's existing test setup (':' is perfectly usable as a path separator if you are running tests in a MSYS2 shell).

closed 2022-03-25 by martinwhitaker

DavidC-75, 2022-03-26:
> in the Windows specific code

I hate mixing things that have nothing to do together... but while you're in the middle of fixing Windows-specific issues, there is an easy one to fix in `driver-vpi/main.c` (see issues/602)

martinwhitaker, 2022-03-26:
Bear in mind that (normally) every bug fix requires a regression test; also I am not a Windows user, so fixing Windows-specific bugs is a painful process.

DavidC-75, 2022-03-26:
> I am not a Windows user, so fixing Windows-specific bugs is a painful process

I appreciate that and I want to thank you all the more for what you are doing !

I only wanted to highlight what might be a low-hanging fruit.... (issues/602 actually reports two different issues - there is a very easy one to fix in `driver-vpi/main.c`, and the other one is clearly more involved, but I'm not suggesting you should look at that too).