steveicarus/iverilog #612: Compile fails with current commit

jaminel, 2022-02-13:
I tried to compile latest commit on Fedora 35 and get errors about undefined references.

Thanks,
-Rad

> make
g++ -rdynamic   -o ivl main.o async.o design_dump.o discipline.o dup_expr.o elaborate.o elab_expr.o elaborate_analog.o elab_lval.o elab_net.o elab_scope.o elab_sig.o elab_sig_analog.o elab_type.o emit.o eval.o eval_attrib.o eval_tree.o expr_synth.o functor.o lexor.o lexor_keyword.o link_const.o load_module.o netlist.o netmisc.o nettypes.o net_analog.o net_assign.o net_design.o netclass.o netdarray.o netenum.o netparray.o netqueue.o netscalar.o netstruct.o netvector.o make_ivl_type.o net_event.o net_expr.o net_func.o net_func_eval.o net_link.o net_modulo.o net_nex_input.o net_nex_output.o net_proc.o net_scope.o net_tran.o net_udp.o pad_to_width.o parse.o parse_misc.o pform.o pform_analog.o pform_disciplines.o pform_dump.o pform_package.o pform_pclass.o pform_class_type.o pform_string_type.o pform_struct_type.o pform_types.o symbol_search.o sync.o sys_funcs.o verinum.o verireal.o vpi_modules.o target.o Attrib.o HName.o Module.o PClass.o PDelays.o PEvent.o PExpr.o PFunction.o PGate.o PGenerate.o PModport.o PNamedItem.o PPackage.o PScope.o PSpec.o PTask.o PUdp.o PWire.o Statement.o AStatement.o LineInfo.o StringHeap.o cprop.o exposenodes.o nodangle.o synth.o synth2.o syn-rules.o t-dll.o t-dll-api.o t-dll-expr.o t-dll-proc.o t-dll-analog.o -ldl
/usr/bin/ld: eval.o: in function `PEFNumber::eval_const(Design*, NetScope*) const':
/project/verilog/iverilog/eval.cc:222: undefined reference to `verireal::as_long(int) const'
/usr/bin/ld: netmisc.o: in function `pad_to_width(NetExpr*, unsigned int, LineInfo const&)':
/project/verilog/iverilog/netmisc.h:170: undefined reference to `pad_to_width(NetExpr*, unsigned int, bool, LineInfo const&)'
/usr/bin/ld: /project/verilog/iverilog/netmisc.h:170: undefined reference to `pad_to_width(NetExpr*, unsigned int, bool, LineInfo const&)'
/usr/bin/ld: /project/verilog/iverilog/netmisc.h:170: undefined reference to `pad_to_width(NetExpr*, unsigned int, bool, LineInfo const&)'
/usr/bin/ld: /project/verilog/iverilog/netmisc.h:170: undefined reference to `pad_to_width(NetExpr*, unsigned int, bool, LineInfo const&)'
/usr/bin/ld: /project/verilog/iverilog/netmisc.h:170: undefined reference to `pad_to_width(NetExpr*, unsigned int, bool, LineInfo const&)'
/usr/bin/ld: netmisc.o:/project/verilog/iverilog/netmisc.h:170: more undefined references to `pad_to_width(NetExpr*, unsigned int, bool, LineInfo const&)' follow
/usr/bin/ld: netmisc.o: in function `eval_as_long(long&, NetExpr const*)':
/project/verilog/iverilog/netmisc.cc:1189: undefined reference to `verireal::as_long(int) const'
/usr/bin/ld: netmisc.o: in function `pad_to_width(NetExpr*, unsigned int, LineInfo const&)':
/project/verilog/iverilog/netmisc.h:170: undefined reference to `pad_to_width(NetExpr*, unsigned int, bool, LineInfo const&)'
/usr/bin/ld: /project/verilog/iverilog/netmisc.h:170: undefined reference to `pad_to_width(NetExpr*, unsigned int, bool, LineInfo const&)'
/usr/bin/ld: /project/verilog/iverilog/netmisc.h:170: undefined reference to `pad_to_width(NetExpr*, unsigned int, bool, LineInfo const&)'
/usr/bin/ld: /project/verilog/iverilog/netmisc.h:170: undefined reference to `pad_to_width(NetExpr*, unsigned int, bool, LineInfo const&)'
/usr/bin/ld: net_func_eval.o: in function `pad_to_width(NetExpr*, unsigned int, LineInfo const&)':
/project/verilog/iverilog/netmisc.h:170: undefined reference to `pad_to_width(NetExpr*, unsigned int, bool, LineInfo const&)'
/usr/bin/ld: net_proc.o:/project/verilog/iverilog/netmisc.h:170: more undefined references to `pad_to_width(NetExpr*, unsigned int, bool, LineInfo const&)' follow
/usr/bin/ld: Statement.o:(.rodata._ZTV6PBlock[_ZTV6PBlock]+0x80): undefined reference to `non-virtual thunk to PBlock::dump(std::ostream&, unsigned int) const'
/usr/bin/ld: Statement.o:(.rodata._ZTV6PBlock[_ZTV6PBlock]+0x88): undefined reference to `non-virtual thunk to PBlock::elaborate(Design*, NetScope*) const'
/usr/bin/ld: Statement.o:(.rodata._ZTV6PBlock[_ZTV6PBlock]+0x90): undefined reference to `non-virtual thunk to PBlock::elaborate_scope(Design*, NetScope*) const'
/usr/bin/ld: Statement.o:(.rodata._ZTV6PBlock[_ZTV6PBlock]+0x98): undefined reference to `non-virtual thunk to PBlock::elaborate_sig(Design*, NetScope*) const'
collect2: error: ld returned 1 exit status

> git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean

larsclausen, 2022-02-13:
This looks like you might have some old .o files that did not get rebuilt. Have you tried a `make clean; make`?

steveicarus, 2022-02-13:
I just did a pull and "make", and it works here. Then to verify, I did another "make clean; make", and that works too. So I suspect Lars is correct, you have some lingering object files and a "make clean" will probably fix it for you. Let me know so I can close this report.

closed 2022-02-26 by steveicarus