steveicarus/iverilog #621: accessing 2D array defined through generate command

etnapoli, 2022-02-21:
I have a module that generates a memory with a nested generate command. 
The final array is a 2D array with 32x4 elements each one on 8 bits.
I see that this array is not automatically dumped using $dumpvars.
I the assigned the elements to a wire and dumped them. 
The results are tricky.  The varius elements are visualized correctly with the execption of the element with [0,0] index.

Below the example of the code I put in the TB.

// definition of the 2D array that accesses to the internal hierarchy
wire [31:0] MEM [31:0];
genvar ii;
generate
for (ii=0;ii<32;ii=ii+1)
	assign MEM[ii] = {UUT.Mycirc.BRAM_GEN[0].BYTE_BRAM_GEN[3].byte_ram[ii],
				UUT.Mycirc.BRAM_GEN[0].BYTE_BRAM_GEN[2].byte_ram[ii],
				UUT.Mycirc.BRAM_GEN[0].BYTE_BRAM_GEN[1].byte_ram[ii],
				UUT.Mycirc.BRAM_GEN[0].BYTE_BRAM_GEN[0].byte_ram[ii]};
endgenerate

// code for the initial block
initial
begin
$dumpfile("waveform.vcd");
$dumpvars;
for (idx = 0; idx < 32; idx = idx + 1) $dumpvars(0, MEM[idx]);

Everything works fine with the MEM elements from 1 to 31. They are visualized and when I chenge the values they behave accordingly.
But when I work with the element MEM[0] only the higher 16bits are changed and the other ones remain xxxxxx.
It looks like there is problem in conncting the MEM0] with the elements:
UUT.Mycirc.BRAM_GEN[0].BYTE_BRAM_GEN[1].byte_ram[0],
UUT.Mycirc.BRAM_GEN[0].BYTE_BRAM_GEN[0].byte_ram[0]

* It is not a visualization probelm. I checke the vcd file visualized by GTKwave and it stores the wrong result.
* the same code is simulate in Mdelsim and provides correct results also for the MEM0] element.

Any hint?

Thx

larsclausen, 2022-02-22:
Hi,

Can you try to provide a minimum working example that reproduces the issue? I've tried the below and the issue did not reproduce

```SystemVerilog
module test2;
reg clk = 0;

always #10 clk <= ~clk;

generate
  genvar a, b;
  for (a = 0; a < 4; a = a + 1) begin: x
    for (b = 0; b < 32; b = b + 1) begin: y
      reg [7:0] z = 0;
      always @(posedge clk)
        z <= z + 1'b1;
    end
  end
endgenerate
endmodule

module test;

test2 test2();

wire [31:0] MEM [31:0];
genvar ii;
generate
for (ii=0;ii<32;ii=ii+1)
assign MEM[ii] = {test2.x[3].y[ii].z, test2.x[2].y[ii].z, test2.x[1].y[ii].z,test2.x[0].y[ii].z};
endgenerate
integer idx;
// code for the initial block
initial
begin
$dumpfile("waveform.vcd");
$dumpvars;
for (idx = 0; idx < 32; idx = idx + 1) $dumpvars(0, MEM[idx]);
#1000;
$finish;
end
endmodule
```

etnapoli, 2022-02-23:
Hi,
 thank you for taking the time to look at the issue.

In attach a working example.
I run it from command line with:
iverilog *.v
vvp a.out
gtkwave waveform.vcd view.gtkw

You can see that the lower part of MEM[0] is xx.
I can tell you that some random changes that I trie remove the issue:
1)  if the wait_for_level task is NOT set to 'automatic' the issue is
resolved
2) if the line 152 is removed
"assign
mybyte=UUT.pattern_find_AXI_full_v1_0_S00_AXI_inst.BRAM_GEN[0].BYTE_BRAM_GEN[0].byte_ram[0];"
the issue resolved.

Thx,
Ettore

-----------------------------------------------------

Prof. Ettore Napoli

Dept. of Information and Electrical Engineering and Applied Mathematics

Università degli Studi di Salerno

Via Giovanni Paolo II, 132 - 84084 Fisciano (SA)

Room: FSTEC-05P02012

Tel: +39 089964207

https://www.diem.unisa.it/

-----------------------------------------------------

etnapoli, 2022-02-24:
I try to attach here one working example of the issue
[pattern_find_AXI_full_v1_0.v.txt](https://github.com/steveicarus/iverilog/files/8134021/pattern_find_AXI_full_v1_0.v.txt)
[tb_ip.v.txt](https://github.com/steveicarus/iverilog/files/8134
[view.gtkw.txt](https://github.com/steveicarus/iverilog/files/8134026/view.gtkw.txt)
024/tb_ip.v.txt)

[pattern_find_AXI_full_v1_0_S00_AXI.v.txt](https://github.com/steveicarus/iverilog/files/8134022/pattern_find_AXI_full_v1_0_S00_AXI.v.txt)

.

martinwhitaker, 2022-03-19:
I can reproduce the fault with the supplied example. The write to the memory byte is working OK, but changes are not propagating through the .array/port (i.e. this isn't a waveform dumper problem). Given that changing an unrelated task from automatic to static fixes (or hides) the problem, this looks like a memory corruption bug in vvp, but we'll need a much smaller test case to track it down. I don't have time to work on that at the moment.

caryr, 2022-03-19:
Depending on the type of memory issue valgrind can be helpful for this and I try to keep vvp clean for just this reason.

martinwhitaker, 2022-03-20:
Here is a much reduced test case that reproduces the bug: [br_gh621.v.txt](https://github.com/steveicarus/iverilog/files/8311626/br_gh621.v.txt)
Changing the body of the `wait_for_wready` task to be an unnamed block also avoids the bug. I suspect the automatic context is not being handled correctly when forking to the nested scope.

martinwhitaker, 2022-03-21:
Fixed in both the master and v11 branches.

closed 2022-03-21 by martinwhitaker

etnapoli, 2022-03-22:
Thank you for solving the bug.
How do I get the new version?

I run MSYS2 and updated iverilog (pacman -Syuu) but it seems that I  still have the same version.

This is what I see in MSYS
mingw64/mingw-w64-x86_64-iverilog 1~11.0.r8956.gf705e7b6-2 (mingw-w64-x86_64-eda) [installed]

and this what I get from the command prompt
Verilog version 12.0 (devel) (s20150603-1179-gf705e7b6)

Thx