steveicarus/iverilog #622: vvp: how to `push` into an instance in a `generate` block?

FriendFX, 2022-02-24:
Sorry if this has been asked before, but I couldn't find any information about using `push` within an interactive `vvp` session to change the scope to an instance which is inside a `generate` block. I am using 

 * Icarus Verilog version 10.1 (stable) ()
 * Icarus Verilog Preprocessor version 10.1 (stable) ()
 * Icarus Verilog Parser/Elaborator version 10.1 (stable) ()
 * vvp.tgt: Icarus Verilog VVP Code Generator 10.1 (stable) ()

Here's some code:

```
// test_gen_push.v
`timescale 1ns/1ps
module ref_clk_gen (output reg ref_clk = 1'b0);
  always @(ref_clk) ref_clk <= #0.5 ~ref_clk;
endmodule

module clkdiv (
    input clk,
    input rstn,
    output reg clk2
  );
  always @ (posedge clk or negedge rstn)
    if (!rstn) clk2 <= 1'b0;
    else       clk2 <= ~clk2;
endmodule

module test_gen_push_tb;
  wire ref_clk;
  reg rstn;

  ref_clk_gen ref_clk_gen (.ref_clk(ref_clk));
   
  localparam INSTANCES = 5;  
  wire [INSTANCES-1:0] clk;
  wire [INSTANCES-1:0] clk2;
  wire out_clk = clk2[INSTANCES-1];
  assign clk[0] = ref_clk;
  
  genvar i;
  generate
    for (i=0; i<INSTANCES; i=i+1) begin: forgen
      clkdiv  divinst (
        .rstn(rstn),
        .clk(clk[i]),
        .clk2(clk2[i])
      );
      if (i>0) assign clk[i] = clk2[i-1];
    end
  endgenerate
  
  initial begin
    $dumpfile ("test_gen_push.vcd");
    $dumpvars(0);
    rstn = 1'b0;
    #10; rstn = 1'b1;
    #50; $stop;
  end
endmodule
```
Command line:
```
$ iverilog -o test_gen_push.o ../../bench/test_gen_push.v && vvp -i test_gen_push.o
VCD info: dumpfile test_gen_push.vcd opened for output.
** VVP Stop(0) **
** Flushing output streams.
** Current simulation time is 60000 ticks.
> ls
1 items in this scope:
module  : test_gen_push_tb
> push test_gen_push_tb
> ls
12 items in this scope:
param   : INSTANCES
reg     : rstn
generate: forgen[0]
generate: forgen[1]
generate: forgen[2]
generate: forgen[3]
generate: forgen[4]
module  : ref_clk_gen
net     : ref_clk
net     : out_clk
net     : clk2[4:0]
net     : clk[4:0]
> push forgen[2]
Scope forgen[2] not found.
> push forgen[2].divinst
Scope forgen[2].divinst not found.
```
Is it possible to enter the `test_gen_push_tb.forgen[2]` scope, or at least `test_gen_push_tb.forgen[2].inst`? How?

steveicarus, 2022-02-26:
That appears to be an honest-to-goodness bug. I will take a closer look at it.

steveicarus, 2022-02-26:
I've merged into master a fix for this. I don't forsee porting the patch to v10 branches, but I probably will port it to the v11 branch.

PR #624 (2022-02-26): Add more types to vvp-stop
Several scope types were not understood by the push command, and
a few types were not displayed reasonably. Flesh these out.

This resolves #622.
closed 2022-02-26 by steveicarus