steveicarus/iverilog #638: Can't parse long verilog attributes

lerwys, 2022-03-11:
When working with a generated vendor module, it uses long attribute lines with line continuation marks, like the attached file [long_attribute_module.txt](https://github.com/steveicarus/iverilog/files/8235172/long_attribute_module.txt).

That confuses iverilog parsing.

```
$ iverilog long_attribute_module.txt 
long_attribute_module.txt:1: Missing close quote of string.
long_attribute_module.txt:2: syntax error
I give up
```

If I try to remove the line continuation marks like the attached file [long_attribute_module2.txt](https://github.com/steveicarus/iverilog/files/8235229/long_attribute_module2.txt), the line is so big that gets rejected.

```
$ iverilog long_attribute_module2.txt
input buffer overflow, can't enlarge buffer because scanner uses REJECT
```

Could we maybe increase the preprocessor limit or support the line continuation syntax with "\\" marks?

Thank you!

ldoolitt, 2022-03-11:
The syntax error is present in 11.0 (at least the version shipped in Debian Bullseye), but already fixed in current git. \o/
The buffer overflow is still there in current git.

lerwys, 2022-03-11:
Oh right. I forgot to paste the version I'm using:

lrusso@lrussoPC:~$ iverilog -v
Icarus Verilog version 10.3 (stable) ()

Good to know it's fixed in the next version @ldoolitt ! 

Thanks!

martinwhitaker, 2022-03-16:
The problem with increasing the buffer size is how far do you go? The default for flex is 16KB. The attached example would require 64KB. But with automatically generated code, who knows how high it could go.

A better solution would be to eliminate the use of REJECT so the lexical analyser could expand its input buffer on demand.

martinwhitaker, 2024-01-08:
I'll label this as a bug, but I don't see an easy way to fix it in general.