steveicarus/iverilog #646: CR (carriage return) does not get parsed in strings

norandomtechie, 2022-03-15:
MVE is as follows:

```
module main;
    logic [7*8-1:0] s;
    initial begin
        s = "hello\r\n";
        $display(s);
    end
endmodule
```

Saved to `test.sv`, and compiled with `iverilog -g2012 test.sv -o out`, and executed with `vvp out`.  The iverilog used is the latest commit to the master branch.

The program should have printed `hello`, but instead printed `hellor`.  It appears that \r is being translated to the regular letter "r", not the CR character (0x13).

caryr, 2022-03-15:
\r is not a valid SystemVerilog escape sequence.

> Table 5-1—Specifying special characters in string literals
> Escape string Character produced by escape string
> \n Newline character
> \t Tab character 
> \\\\ \ character
> \\\" " character
> \v vertical tab
> \f form feed 
> \a bell
> \ddd A character specified in 1 to 3 octal_digits (see Syntax 5-2). If fewer than three 
> characters are used, the following character shall not be an octal_digit. Implementations 
> may issue an error if the character represented is greater than \377. It shall be illegal for 
> an octal_digit in an escape sequence to be an x_digit or a z_digit (see Syntax 5-2). 
> \xdd A character specified in 1 to 2 hex_digits (see Syntax 5-2). If only one digit is used, the 
> following character shall not be a hex_digit. It shall be illegal for a hex_digit in an 
> escape sequence to be an x_digit or a z_digit (see Syntax 5-2).

martinwhitaker, 2022-03-15:
As Cary says. All other simulators I tried behaved the same. Closing as invalid.

closed 2022-03-15 by martinwhitaker