steveicarus/iverilog #648: No warning given for inferred latch

hughperkins, 2022-03-15:
Eg if I do:

```
module test(input a, output reg b);
    always @(*) begin
        if(a) b = ~b;
    end
endmodule
```
then compile with iverilog:

What I expect:

```
warning: inferred latch for variable `a`
```

What I actually see:

```
$ iverilog /tmp/test.sv 
$
```
(i.e. no warning currently given. cf eq Quartus gives such a warning)

closed 2022-03-15 by steveicarus

martinwhitaker, 2022-03-16:
You can read the man page (as mentioned in section 1.0 of the README.txt file) or follow the Wiki link at the top of this (and every other) GitHub page. I don't know how we can make it more obvious than that.

`iverilog -h` gives a summary of the options, and again suggests reading the man page.

hughperkins, 2022-03-16:
Ah, `man iverilog` does give doc. Thanks.

> iverilog -h gives a summary of the options, and again suggests reading the man page.

That's fair. I'm on a Mac, and almost never use man pages, don't have the reflex. Usually just google, eg `iverilog`, which gets me to http://iverilog.icarus.com/

> or follow the Wiki link at the top of this (and every other) GitHub page. I don't know how we can make it more obvious than that.

I dont see this link. Do you mind providing the URL here?  Here is what I see in the various pages I've looked at:

iverilog main page, from googling 'iverilog'. I see instructions on instalation, how to get from git, links to other projects (verilator etc), but I dont see any link to documentation, or wiki page?

[image]

Top of github page:

[image]

I see a link on the top right, which takes me to the iverilog home page, that I pasted above. I dont see any links to documentaiton, or wiki?

Top of github readme:

[image]

(note that it's quite a long scroll to get to github readme, so I think you might consider moving the .cc and .h files into a `src` sub-folder, to keep the github main page relatively compact)

So, in the top of the readme,  I see another link to the iverliog main page, and then description of how to build/install iverilog, but I don't see either documentaiotn of parameters, or a link to a documentaiton page, or a wiki page?

(This discussion should probably be a separate isssue really. shall I move this to a separate issue? Happy to do that)

martinwhitaker, 2022-03-16:
> I dont see this link. Do you mind providing the URL here? Here is what I see in the various pages I've looked at:
...
> Top of github page:
Look at the tabs: Code,  Issues,  Pull requests, ..., Wiki, ....
The Wiki tab contains the link: http://iverilog.wikia.com/

For other matters, best start a new topic in the Discussions section.

hughperkins, 2022-03-18:
Ok, so, I just spent 4 hours debugging my code, and it was because I hadn't initialized a reg. How do you guys detect regs that haven't been initialized? (I checked verilator, and it also happily compiles my code, without a warning, even with `-Wall`. verible linter rules appear not to check for this. What do you guys do in practice?

hughperkins, 2022-03-18:
(Hmmm, actually, the new variable uninitialized issue was with a flip-flop, not an inferred latch; separate issue; basically needs me to either change all `if` statements to `case` with a default that catches `x`, or else put `assert(~$isunknown(some_var))` before any `if` statements. Issue is taht behavioral `if`s send an `x` to the `else`, whereas gate-level `if`s, whihc are now gates, propagate `x`s, and give differet results).