steveicarus/iverilog #649: Bug in variable name for 'not all bits are asynchronously set or reset. This is not currently supported in synthesis'.

hughperkins, 2022-03-15:
Use `iverilog -S -g2012` to compile the following:

```
module test(input clk, input rst, input a, output reg [1:0] state);
    reg [1:0] next_state;
    reg [3:0] pos;
    reg [3:0] next_pos;

    always @(*) begin
        next_state = state;
       if(a) begin
           next_state = '0;
       end
    end
    always @(posedge clk, posedge rst) begin
        if(rst) begin
            state <= '1;
            // pos <= '0;
        end else begin
            state <= next_state;
            pos <= next_pos;
        end
    end
endmodule

```

Gives message `sorry: Not all bits of 'state' are asynchronously set or reset. This is not currently supported in synthesis.`

Turns out it's actually because of pos, not state. uncommenting the line that sets pos to 0s gets rid of the error message.

martinwhitaker, 2022-03-19:
Fixed in both master and v11 branches.

closed 2022-03-19 by martinwhitaker

hughperkins, 2022-03-19:
Awesome! Thank you :)