steveicarus/iverilog #650: Uninformative error `error: Unable to synthesize asynchronous process.`

hughperkins, 2022-03-15:
If I use `iverilog -g2012 -S` to compile the following:

```
module test(input a, output reg b);
    always @(*) begin
        $display("foo");
        b = a;
    end
endmodule
```

What I expect to happen:

either:
- get a message saying cannot synthesize `$display()`, or
- synthesizes, and simply ignores `$display` ( this is what yosys does for example)

What actually happens:

```
/tmp/test.sv:2: error: Unable to synthesize asynchronous process.
1 error(s) in post-elaboration processing.
```

Also, note that the line number doesnt even point to the `$display`. It points to the always. Now, I'm not going to say it's impossible to debug, because my filing this issue proves it's possible to find the issue, but it was tricky to track down, and I feel the error message could be more informative, and the line number too.

hughperkins, 2022-03-15:
(Note: my personal preferred behavior would be to simply silently ignore `$display`, `$strobe`, `$monitor` at synthesis.)

martinwhitaker, 2022-03-16:
Note that the synthesis option has not been actively developed for over a decade. What's there should work, but it's unlikely any effort will be put into improving it. If you want an open source synthesis tool, yosys is the best option I know of.

hughperkins, 2022-03-16:
Yes, so, I use yosys for synthesis. However, I'm using iverilator for simulation. My actual goal for running synthesis on iverilator is not to run syntehsis, but because in another issue I was informed that it is the standard approach to receive 'latch inferred' warnings. See https://github.com/steveicarus/iverilog/issues/648 . Personally, I would *much* prefer to receive such warnings using standard iverilog simulation.

Currently, my workflow for receiving latch warnings is  quite long-winded. It comprises:

- run yosys, synth
- save as gate-level netlist verilog
- open the gate-level netlist verilog, and search for `always` => if there is an always, so far it has always meant there was an inferred latch, i.e. one of my regs was not initialized at the start of my combinatorial always block

Using eg Quartus, one gets warned immediately, during simulation, e.g. you can try quartus at https://hdlbits.01xz.net/wiki/Problem_sets (Quartus is not opensource, so I'm not using it, but I do like that it warns me about inferred latches).