steveicarus/iverilog #660: `assert: elab_expr.cc:2201: failed assertion base_index.size()+1 == net->packed_dimensions()` when linting systemverilog file

xsun2001, 2022-03-24:
I'm trying to use iverilog to lint my systemverilog code. But i got an assert error.

Env: macOS 12.3 on M1 Max MacBook Pro
Ver: `icarus-verilog` stable 11.0 from homebrew

```
> iverilog -g2012 -Wall -i -Y .sv -y ./src -I ./src/include -t null src/test.sv
test.sv:120: assert: elab_expr.cc:2201: failed assertion base_index.size()+1 == net->packed_dimensions()
sh: line 1: 10739 Done                    /opt/homebrew/Cellar/icarus-verilog/11.0/lib/ivl/ivlpp -L -Wredef-chg -F"/var/folders/nr/9ld3b5qs3jg_dgvj709_0g1w0000gn/T//ivrlg2ac18d37" -f"/var/folders/nr/9ld3b5qs3jg_dgvj709_0g1w0000gn/T//ivrlgac18d37" -p"/var/folders/nr/9ld3b5qs3jg_dgvj709_0g1w0000gn/T//ivrliac18d37"
     10740 Abort trap: 6           | /opt/homebrew/Cellar/icarus-verilog/11.0/lib/ivl/ivl -C"/var/folders/nr/9ld3b5qs3jg_dgvj709_0g1w0000gn/T//ivrlhac18d37" -C"/opt/homebrew/Cellar/icarus-verilog/11.0/lib/ivl/null.conf" -- -
```

martinwhitaker, 2022-03-25:
Please could you provide a small test case that allows us to reproduce the bug.

martinwhitaker, 2024-01-09:
No answer after more than 1 year, so closing. Please reopen if you can provide a way to reproduce the issue.

closed 2024-01-09 by martinwhitaker