steveicarus/iverilog #661: 'urandom_range' unable to support 64-bit range

AmeyaVS, 2022-03-26:
It seems the current implementation of `urandom_range` is restricted to less than 32-bit integer range.

While trying use SystemVerilog functions `urandom_range` I hit an issue while trying to generate stimulus for a testbench.

Here is a small code which reproduces the issue `unrandom_bug.v`:

```sv
module urandom_bug;

reg [64:0] something;

initial begin
	repeat(100) begin
`ifdef MAX_32_RANGE
		something = $urandom_range(64'hffff_ffff, 64'h0);
`else
		something = $urandom_range(64'hfff_ffff, 64'h0);
`endif
		$display("%h", something);
		#10;
	end
end

endmodule
```

**Scenario 1**
When trying with the `urandom_range` with `maxval` set to `((2^32) - 1)` I get only zeros as values.

**_Build Command_**
```sh
iverilog -g2012 -DMAX_32_RANGE urandom_bug.v -o urandom_bug_max.out
```
**_Output:_**
```sh
$ ./urandom_bug_max.out 
00000000000000000
00000000000000000
00000000000000000
00000000000000000
00000000000000000
00000000000000000
00000000000000000
00000000000000000
.....
```

**Scenario 2**
**_Build Command_**
```sh
iverilog -g2012 urandom_bug.v -o urandom_bug_limit.out
```

**_Output_**
```sh
$ ./urandom_bug_limit.out 
00000000009215352
000000000040895e8
00000000000484d60
000000000031f0566
000000000086b97b0
.....
```

Any value to `maxval >= ((2^31)` to `urandom_range`  triggers the same behavior as in `Scenario 1`.

martinwhitaker, 2022-03-27:
This bug report is confusing as currently written. Your code demonstrates that Icarus does not handle the full unsigned 32-bit range, which is indeed a bug. But the bug title and the example code indicate you want 64 bit values, and that is not valid - the arguments and return value for urandom_range are `int unsigned`, which is 32 bits.

AmeyaVS, 2022-03-27:
Seems to be my bad.
I think I missed the `int unsigned` return type.

Then I will update the title and the text to reflect the 32-bit range limitation in the current implementation.

martinwhitaker, 2022-03-27:
@AmeyaVS , thanks for the update.

Looking at the code in sys_random.c, I see much use of `long` and `unsigned long`, which will be 64 bit values on a 64-bit Unix system (but 32 bits on a 64-bit Windows system). From https://accellera.mantishub.io/view.php?id=1136, I think this is wrong, and we should be using `int32_t` and `uint32_t` to ensure 32 values are used on all platforms. @caryr and @steveicarus , do you agree?

martinwhitaker, 2022-04-03:
Fixed in the master branch. I won't backport this to the v11 branch until it's had wider testing.

AmeyaVS, 2022-04-16:
Can confirm the changes works.
Do I close the issue for now or want me to keep the issue open to get wider response on the changes?

AmeyaVS, 2023-03-07:
Closing the issue as fixes are available on the main branch.