steveicarus/iverilog #672: "fork" blocks re-triggering of always block

mole99, 2022-04-06:
In this module a new process should be started on every `posedge`:

```
module top;

    logic clk = 0;
    int cnt = 0;

    always @(posedge clk) begin
        fork begin
            #(10*2); // Wait 10 clock periods
            $display("End of this process");
            cnt++;
        end
        join_none
    end

    initial begin
        $display("Starting test!");
        #(100*2); // Wait 100 clock periods
        $display("cnt = %d", cnt);
        $finish;
    end
    
    always begin
        #1 clk = ~clk;
    end

endmodule
```

The output is the following:

```
iverilog -o test -s top -g2012 test.sv
vvp test
Starting test!
End of this process
End of this process
End of this process
End of this process
End of this process
End of this process
End of this process
End of this process
End of this process
cnt =           9
iverilog_test.sv:18: $finish called at 200 (1s)
```

The value of `cnt` at the end of execution is 9 instead of the expected 90. This means when the always block was triggered it was only possible to trigger it again when the process inside fork has finished.

Running the same example on Xcelium xrun 20.09 returns the value 90 for `cnt`.

martinwhitaker, 2022-04-06:
It looks like the tgt-vvp code generator is optimising away the fork because it only contains one process. That probably dates back to before SystemVerilog - it would be a valid thing to do for a straight fork-join, but not for a fork-join_none.

This workaround gives the expected behaviour:
````
    always @(posedge clk) begin
        fork begin
            #(10*2); // Wait 10 clock periods
            $display("End of this process");
            cnt++;
        end
        #0;
        join_none
    end
````

martinwhitaker, 2022-04-15:
Fixed in both the master and v11 branches.

closed 2022-04-15 by martinwhitaker