steveicarus/iverilog #674: Elaborator segfault during function evaluation

dieterss, 2022-04-10:
Hello,

There is a segfault in the elaborator when it attempts to evaluate constant functions which access arrays out-of-bounds, as if some bounds check where missing. Compiling the code below (iverilog test.v) segfaults every time.

I checked version 0.8.13 (debian stable) and the latest git: both segfault on this code.

```verilog
module test;

function func;
input x;

integer i;
reg [5 - 1:0] c[1:0];
begin
        for (i = 0; i < 10; i = i + 1)
                c[-i] = 123;

        func = 0;
end
endfunction

localparam X = func(0);

endmodule
```

martinwhitaker, 2022-04-10:
Fixed in both the master and v11 branches.

closed 2022-04-10 by martinwhitaker

dieterss, 2022-04-12:
Thank you for the fix and the fast turn around time!