steveicarus/iverilog #678: Strict expression width doesn't apply to event triggers

zachjs, 2022-04-12:
```SystemVerilog
module top;
    reg x, y;
    always @(x + y)
        $display("%0d %b %b", $time, x, y);
    initial begin
        #1 {x, y} = 2'b00;
        #1 {x, y} = 2'b01;
        #1 {x, y} = 2'b10;
        #1 {x, y} = 2'b11;
        #1 {x, y} = 2'b00;
        #1 {x, y} = 2'b10;
    end
endmodule
```

When compiled with `-gstrict-expr-width`, I would expect the `always` block to only be triggered when the parity changes. Indeed, when using `^` rather than `+` , iverilog simulates this correctly.

Results in iverilog, only with `+`, even with `-gstrict-expr-width`:
```
1 0 0
2 0 1
3 1 0
4 1 1
5 0 0
6 1 0
```

Results in other simulators, using either `+` or `^`:
```
1 0 0
2 0 1
4 1 1
6 1 0
```

This appears to deal with event trigger evaluation in general, rather than `event_expression`s alone, as the following modification still results in the errant results.
```SystemVerilog
    ...
    wire z;
    assign z = x + y;
    always @(z)
    ...
```

zachjs, 2022-04-13:
Thank you for explaining that! While I cannot dispute the behavior for concatenations, essentially equivalent behavior can be achieved without a concatenation, producing a single event for each assignment.

```SystemVerilog
module top;
    reg [1:0] i;
    wire o;
    assign o = i[0] + i[1];
    always @(o)
        $display("%0d %b %b", $time, i, o);
    initial begin
        #1 i = 2'b00;
        #1 i = 2'b01;
        #1 i = 2'b10;
        #1 i = 2'b11;
        #1 i = 2'b00;
        #1 i = 2'b10;
    end
endmodule
```

If the sum is indirected through a size-2 net, the extra triggers go away. If `f`'s dimension is removed, the extra triggers remain.
```SystemVerilog
    ...
    wire [1:0] f;
    assign f = i[0] + i[1];
    assign o = f;
    ...
```

martinwhitaker, 2022-04-13:
Yes, that is https://sourceforge.net/p/iverilog/bugs/772/. @caryr, do you think you'll ever have time to complete your rework of expression evaluation?

martinwhitaker, 2024-01-08:
Whilst the behaviour with the original example is probably legal, the behaviour with the second example is not, so I'll label this as a bug. Like the old bugs on SourceForge, I think the only plausible way to fix this is to change vvp to use a pull model for expression evaluation.