steveicarus/iverilog #687: Misleading warning when for loop index variable is of type logic inside always block

mole99, 2022-04-15:
I have this simple module:

```
module top;
    logic [9:0] pipe = 0;
    logic [4:0] i;
    logic clk = 0;

    always #1 clk = ~clk;

    always_ff @(posedge clk) begin
        for (i=0; i<9; i++) begin
            pipe[i+1] <= pipe[i];
        end
        pipe[0] <= pipe[9];
    end

    initial begin
        pipe[0] = 1'b1;
        
        for (int j=0; j<10; j++) begin
            $display(pipe[9]);
            #2;
        end
        
        $finish;
    end
endmodule
```

When this code is run, I get the expected output, but also a warning concerning the index variable `i` of the for loop.
As far as I see this warning is misleading. The index i is used in the condition expression (`i<9`).

```
iverilog -o Vtop.vvp -g2012 top.sv 
top.sv:9: warning: A for statement must use the index (i) in the condition expression to be synthesized in an always_ff process.
vvp Vtop.vvp
0
0
0
0
0
0
0
0
0
1
top.sv:24: $finish called at 20 (1s)
```

Then I changed `logic [4:0] i;` to just `int i;`. This seems to be the proper way to do this and I no longer get a warning:

```
iverilog -o Vtop.vvp -g2012 top.sv 
vvp Vtop.vvp
0
0
0
0
0
0
0
0
0
1
top.sv:24: $finish called at 20 (1s)
```

martinwhitaker, 2022-04-15:
The more general workaround is to make the LHS and RHS of the comparison the same size, e.g. with your original code:
````
        for (i=0; i<5'd9; i++) begin
````
The bug occurs when the compiler needs to extend the value of the loop variable to match the expression width.

martinwhitaker, 2024-01-07:
Fixed in the master branch.

closed 2024-01-07 by martinwhitaker