steveicarus/iverilog #703: always block in generate loop error

qian-gu, 2022-04-28:
Hi, I get a error message saying `error: Theses bit are already driven.`  when comiple following code:

```
logic [7 : 0] dout;
logic [7 : 0] sel;

for (genvar i = 0; i < 8; i++) begin
  if (i == 0) begin

    assign dout[i] = 1'b0;

  end else if (i == 1) begin

    assign dout[i] = 1'b1;

  end else begin

    // using always block reports error
    always @(*) begin
        if (sel[i]) dout[i] = 1'b1;
        else dout[i] = 1'b0;
    end

  end
end
```

But if I replace the always block with assign statement, the error message disapears,

```
logic [7 : 0] dout;
logic [7 : 0] sel;

for (genvar i = 0; i < 8; i++) begin
  if (i == 0) begin

    assign dout[i] = 1'b0;

  end else if (i == 1) begin

    assign dout[i] = 1'b1;

  end else begin

    assign dout[i] = sel[i] ? 1'b1 : 1'b0;

  end
end
```

so is this a bug or intend result of some consideration?

martinwhitaker, 2022-04-30:
It's a bug. A simpler test case is
````
module test();

logic       a;
logic [1:0] v;

assign v[0] = 1'b0;

always @* begin
  if (a)
    v[1] = 1'b1;
  else
    v[1] = 1'b0;
end

endmodule
````
Because `v` is partially driven by a continuous assignment, the compiler is checking that it is not driven by multiple assignments. But that should only be done for continuous assignments, not for procedural assignments..

martinwhitaker, 2024-02-09:
This was fixed by commit d4759e02aaf88a920ee48849056e7d133a44f282.

closed 2024-02-09 by martinwhitaker