steveicarus/iverilog #710: `VCD warning: $dumpvars: Package ($unit) is not dumpable with VCD` if compiled with `-g2012`

goekce, 2022-05-07:
I use the latest snapshot. When I compile the following module with `-g2012` I get a VCD warning.

`a.v`:
```verilog
module m;

logic s;

initial begin
	$dumpfile("signals.vcd");
	$dumpvars();
	s = 1;
end
endmodule
```

```
$ iverilog -g2012 a.v
$ ./a.out
VCD info: dumpfile signals.vcd opened for output.
VCD warning: $dumpvars: Package ($unit) is not dumpable with VCD.
```

`signals.vcd` file is readable. I do not get the error if I leave out `-g2012`.

Is this a feature or a bug?

martinwhitaker, 2022-05-07:
It's a feature. SystemVerilog allows you to declare items in the compilation unit scope ($unit). The compilation unit scope is not actually a package, but it has the same characteristics, so is treated as such by the compiler and vvp. The VCD file format has no support for either packages or compilation unit scopes, hence the warning.

It would be nice to suppress the warning in the case that there are no items declared in the compilation unit scope.

goekce, 2022-05-09:
thanks Martin. I will leave this issue there then.

martinwhitaker, 2024-01-07:
Warnings about dumping packages (including the $unit scope) are now suppressed if the package contains no dumpable items.

closed 2024-01-07 by martinwhitaker