steveicarus/iverilog #711: questions when $display("%d",$time) in sub-modules

wytWangLei, 2022-05-09:
When I running testbench and using $display("%d",$time) in sub-modules, vvp simulation output always shows zero. It acts like the $time in sub-modules is assigned when instantiated so it won't changed as simulation processing. Is it a bug or my misuse？

martinwhitaker, 2022-05-09:
I would guess it's a bug in your testbench. This simple example produces the expected output:
````
module sub_module;

initial begin
  repeat (10) #1 $display("%d", $time);
end

endmodule

module tb;

sub_module m();

endmodule
````

wytWangLei, 2022-05-10:
@martinwhitaker Thank you for your reply. Maybe I should describe my question clearly. My testbench file is like this:
```
module tb;
    reg clk;
    reg reset;
    initial begin
        reset = 1;
        clk = 0;
        #10 reset = 0;
        $display("%d",$time);
        #990 $finish;
    end
    always #5 clk = ~clk;

    initial begin
        $dumpfile("wave.vcd");
        $dumpvars(0,tb);
    end
    mips u_mips(
    	.clk   (clk   ),
        .reset (reset )
    );
endmodule
```
while the module 'mips' and other sub-modules are defined in other .v files. Those modules may contain use of $time. And I use the "iverilog -y ./ *.v" command to generate simulation file for vvp. Other output is correct, but $time is always shown as 0.

steveicarus, 2022-05-10:
You will need to post a minimal (yet complete) example that demonstrates your issue.
https://steveicarus.github.io/iverilog/usage/reporting_issues.html

The advance of time is part of the simulation engine, and nothing is able to write to the time. So there is more to what you are showing.

wytWangLei, 2022-05-10:
@steveicarus I really appreciate that you noticed my question. I've found that it's `timescale causing the $time showing zero. Is `timescale not supported by iverilog/vvp?

steveicarus, 2022-05-10:
The timescale directive is definitely supported. Is it possible you set it so that the time being printed is rounded to zero?

wytWangLei, 2022-05-11:
@steveicarus Yes, I found that it's my misuse of iverilog command. I always use `iverilog -y ./ *.v`, which causes other modules using default timescale as 1s/1s. The correct command is like `iverilog -y ./ tb.v`, then the timescale in top-level is correctly passed on. Thank you for your patient help!

steveicarus, 2022-05-15:
Sounds like your problem is resolved, so I'm closing this issue.

closed 2022-05-15 by steveicarus