steveicarus/iverilog #723: Nested comment error (bug)

mkhazraee, 2022-05-27:
After checking out the latest version of the repo (V12), I get a new error compared to the stable v11 tag. The error is from one of the Verilog files and says _error: Possible nested comment_

However it is not a nested comment, It's something like 
/* 
// ...
...
*/

Which is valid for Verilog syntax and is not nested.

mkhazraee, 2022-05-27:
Tracked it down to first change in commit 31a68fcd8a9ceaa51ed1a68b3383a0c144ece15f. Removing that line fixes the issue, as it's too restrictive.

martinwhitaker, 2022-08-28:
I tried to reproduce this with
````
module test;

/*
// hello
world
*/

endmodule
````
but got no error. Can you provide a simple example the demonstrates the issue.

mkhazraee, 2022-11-16:
Hi, I checked a later commit and the issue is gone. Not sure which commit, but it's gone anyways. Thanks

closed 2022-11-16 by mkhazraee