steveicarus/iverilog #732: Assertion failure on specparam declaration of delay value when enclosing module is instantiated more than once.

martinwhitaker, 2022-06-14:
This short example demonstrates the problem:
````
module Buffer(output Y, input A);

  assign Y = A;
             
  specify
    specparam tpdLH = (0.01:0.01:0.01);
    specparam tpdHL = (0.01:0.01:0.01);
  endspecify
endmodule
  
module Buffer1(output Y, input A);
  Buffer inst(.Y(Y), .A(A));
endmodule
  
module Buffer2(output Y, input A);
  Buffer inst(.Y(Y), .A(A));
endmodule
````
resulting in
````
% iverilog test.v 
test.v:6: warning: choosing typ expression.
test.v:7: warning: choosing typ expression.
test.v:7: assert: ../source/net_design.cc:700: failed assertion param_type
````
This only occurs in development. v11 compiles the above example without error.

jayschleicher, 2022-06-15:
I see a similar issue with this bit of Verilog

`module param_tst_one();
   parameter PARAM = 1.25;
endmodule

module param_tst();
   // Needs to be a real number
   localparam PARAM = 1.25;   
//   localparam PARAM = 2;   

   genvar i;
   // Needs to generate at least two instances
   generate for (i=2; i>0; i=i-1) begin
	  param_tst_one tst () ;
	  defparam tst.PARAM = PARAM ;
   end
   endgenerate
endmodule
`
This results in the same error mentioned above. 
I also noticed the parameter had to be a real value to have this issue occur. 
It does not happen in v11 only the development version.

martinwhitaker, 2022-08-24:
Fixed.

closed 2022-08-24 by martinwhitaker