steveicarus/iverilog #734: sorry: constant selects in always_* processes are not currently supported

bjourne, 2022-06-21:
I'm getting this error from this code with iverilog 11.0:
```
    always_comb begin
        if (ac >= {1'b0,y1}) begin
            ac_next = ac - y1;
            {ac_next, q1_next} = {ac_next[WIDTH-1:0], q1, 1'b1};
        end else begin
            {ac_next, q1_next} = {ac, q1} << 1;
        end
    end
```
Is the sorry message a warning or an error and does it change how the code works? As an side, it would be really useful to have a table of what System Verilog features are supported in what version and also what plans you have on adding missing features. Icarus Verilog is still a  great tool though.

martinwhitaker, 2022-06-24:
The full message is
````
sorry: constant selects in always_* processes are not currently supported (all bits will be included).
````
As far as I can see, that means the always_comb process will be triggered by changes in any bits of `ac_next`, not just `ac_next[WIDTH-1:0]`. Only you can tell if that matters in your design, but I would be surprised if it did.

On the subject of documenting what is supported, see issue #206.  Speaking just for myself, I only use standard Verilog, so have no plans to add missing System Verilog features.

bjourne, 2022-06-24:
Thanks. It is very good to know that you are not targeting the full System Verilog spec.

caryr, 2022-06-25:
Note that Martin was speaking for himself. Some of us have significant interest in adding support for various 
SystemVerilog constructs, but for the last number of years I have been completely consumed with designing actual circuits that many reading this comment likely use on a daily basis.

One way to remove this specific limitation is to make the constant select part of a separate assignment to a dedicated variable instead of it being inside the always construct. Another benefit of doing this is it gives the selection a name that could be more descriptive than using the select.

steveicarus, 2022-12-05:
@bjourne Can you attach to this issue a complete Verilog source file that replicates this issue. Your snippet is fine for expressing what the issue is, but we need the code around it (such as declarations) that make this example runnable.

chykon, 2023-07-11:
Here is another example to reproduce this issue:

```systemverilog
module example (
  input logic [7:0] d,
  output logic [3:0] q
);
  always_comb begin
    q = d[3:0];
  end
endmodule

module testbench;
  logic [7:0] d;
  logic [3:0] q;

  example example_instance (.d(d), .q(q));

  initial begin
    d = 255;
    #0;
    $display(q);
  end
endmodule
```

Askartos, 2024-11-24:
Hi
I did find this issue but to my surprise iverilog doesn't seem to be doing what the message states. the constant selections seem  not to include  all bits  of the operand (below my testcase)
```systemverilog
`timescale 1 ns/1 ps
module alu_tb;
  logic [31:0] operand_1=32'd1;
  logic [31:0] operand_2=32'd37;
  logic [31:0] shift_2lsb;
  logic [31:0] shift_5lsb;
  logic [31:0] shift_32bits;
  alu alu_dut(
    .operand_1     (operand_1),
    .operand_2     (operand_2),
    .shift_2lsb(shift_2lsb),
    .shift_5lsb(shift_5lsb),
    .shift_32bits(shift_32bits)
  );
  initial    begin
    $display("operand_1\t= 0x%h",operand_1);
    $display("operand_2\t= 0x%h",operand_2);
    #1;
    $display("shift_2lsb\t= 0x%H",shift_2lsb);
    $display("shift_5lsb\t= 0x%H",shift_5lsb);
    $display("shift_32bits\t= 0x%H",shift_32bits);
    $finish();
  end
endmodule
module alu(
input  logic [31:0] operand_1,
input  logic [31:0] operand_2,
output logic [31:0] shift_2lsb,
output logic [31:0] shift_5lsb,
output logic [31:0] shift_32bits
);
//Arithmetic operations
always_comb begin
      shift_2lsb = operand_1<<operand_2[1:0];
      shift_5lsb = operand_1<<operand_2[4:0];
      shift_32bits = operand_1<<operand_2;
end
endmodule 
```
cmdline :
```bash
>>>iverilog -g2012 alu_tb.sv -o alu.sim
alu_tb.sv:33: sorry: constant selects in always_* processes are not currently supported (all bits will be included).
alu_tb.sv:33: sorry: constant selects in always_* processes are not currently supported (all bits will be included).
 >>>./alu.sim
operand_1	= 0x00000001
operand_2	= 0x00000025
shift_2lsb	= 0x00000002
shift_5lsb	= 0x00000020
shift_32bits	= 0x00000000
>>> iverilog -V
Icarus Verilog version 11.0 (stable) (v11_0-132-gb2f2414f4)

Copyright 1998-2020 Stephen Williams

```
If all bits were included all three operations should have resulted in 0x0

martinwhitaker, 2024-11-24:
@Askartos , see my previous comment. The message is a bit misleading.

BTW, you are using a very old version of Icarus Verilog. Many bugs have been fixed since that was released.

caryr, 2024-11-25:
This message is related to the `always_comb/latch` statement calculating the correct sensitivity list. It currently uses the same scheme as is used for `@*` which does not match what is required for `always_comb/latch`. Specifically, a constant select does not limit the sensitivity to only the bits selected. It does not limit the actual block statements in any way. Since your code would already be sensitive to the entire vector because of the last assignment in the block this has no impact on how your code is simulated.