steveicarus/iverilog #735: support for system verilog

TomO700, 2022-06-25:
Hi,

I was just trying out icarus verilog on edaplayground and it seemed as if basic constructs such as` always_ff, always_comb, logic `are not supported?
May I check if this is the case for icarus in general or is it specifically for the version in cocotb?
Additionally, if icarus does support some system verilog constructs, may I please know where I can check what those are(I have heard it supports a few)

Thanks a lot!

svenka3, 2022-06-25:
I don't know if the EDAPlay version is devel or not. In Devel tree the below code works fine:

iverilog -g2012 alw_comb_ff.sv

```verilog
module test ();
  reg y;
  reg y_sv;

	  bit clk, rst_n;

		bit q;

  always_ff @ (posedge clk) begin
		if (!rst_n) begin
			q <= 0;
		end else begin
			q <= y;
		end
  end

  always_comb begin
    y_sv = 1'b0;
  end
  initial begin
    #1;
    $display("Value of y: %b", y);
    $display("Value of y_sv: %b", y_sv);
  end
endmodule : test
```

caryr, 2022-06-25:
Logic is fully supported, always_comb and always_ff are also supported with some limitations.

We do not have a list of SV features that have been implemented, partially implemented or planned for implementation. The following link may give you some idea about the support, but their tests have some assumptions about constructs that are supported and when these are missing false failures are reported: [SV-Tests](https://chipsalliance.github.io/sv-tests-results/)

The Icarus versions on EDA playground are significantly out of date and should not be used to judge what Icarus does or does not support. Also note that we often recommend people use the development branch for the most complete support. This is likely different than many projects recommendations since the development branch for them is often unstable. For Icarus we try to keep development stable with major changes happening in branches and explicit announcements that development may be unstable for a period of time when changes that could cause instability are integrated.

TomO700, 2022-06-25:
Oh ok. Thanks for the insight.
I am trying to run it without edaplayground but I keep getting the following error, in 3 different versions of cocotb.
CRITICAL Failed to import module test_endian_swapper: No module named 'cocotb.drivers'
Any ideas for fixes?

caryr, 2022-06-26:
That looks like a cocotb setup issue. I believe Icarus is regularly tested to work with cocotb.

closed 2022-06-28 by TomO700

MengXiao92, 2022-12-17:
I met the same issue, do you @svenka3 fix  the issue about the  always_ff and always_comb? 
thanks

svenka3, 2022-12-17:
@MengXiao92 not sure what I had to "fix", I only found that the code snippet in my earlier post worked fine in Icarus..

> In Devel tree the below code works fine: