steveicarus/iverilog #736: simulator inconsistency(nedsted module and flatten f/f)

sj0000e, 2022-06-26:
I test simple gated f/f.
https://www.edaplayground.com/x/mLAH

Rivera,Questa follow convention 
vcs follows standards.
xcelium, and icarus verilog has inconsitent simulatorc
The results not the same depending on where f/f is, and signal path

26000000000000 ps, wr_en: 1 ,flop input : 11    - will be update to new value
26000000000000 ps, wr_en: 1 ,flop output: 00 11 00 11 - before rising edge 
36000000000000 ps, wr_en: 0 ,flop output: 00 11 11 11 - after  rising edge, expect update

martinwhitaker, 2022-06-26:
That is because there is a race in your code. Look at the assignments to `d` and `de`.

sj0000e, 2022-06-26:
I got it. verilog standard allow that sequence.
that is evaluted _some_ previous simulation time.

IEEE Std 1364-2005,  11.3 The stratified event queue
c) Nonblocking assign update events have been evaluated during some previous simulation time, but shall be assigned at this simulation time after all the active and inactive events are processed.

IEEE Std 1800-2017
4.4.2.4 NBA events region
The NBA (nonblocking assignment update) region holds the events to be evaluated after all the Inactive
events are processed.
If events are being executed in the active region set, a nonblocking assignment creates an event in the NBA
region scheduled for the current or a later simulation time.

closed 2022-06-26 by sj0000e