steveicarus/iverilog #747: assert: elab_net.cc:1028: failed assertion sig

bjourne, 2022-08-03:
This code produces an error in iverilog 11.0:

    module sub #(parameter WIDTH=8) (
        input [WIDTH-1:0] x [2:0],
        output [WIDTH-1:0] y
    );
        assign y = 2 * x[2];
    endmodule
    module test #(parameter WIDTH=8) (
        input [WIDTH-1:0] t
    );
        sub #(.WIDTH(WIDTH)) m1(E[0], E[1]);
        sub #(.WIDTH(WIDTH)) m2(E[1], E[2]);
        sub #(.WIDTH(WIDTH)) m3(E[2], E[3]);
    endmodule
    
    $ iverilog -g2012 err.v
    err.v:10: assert: elab_net.cc:1028: failed assertion sig

bjourne, 2022-08-03:
This code produces the error `assert: elaborate.cc:1630: failed assertion rval_net->pin_count() == prts[0]->pin_count()`:

    module sub #(parameter WIDTH=8) (
        input [WIDTH-1:0] x [0:2],
        output [WIDTH-1:0] y [0:2]
    );
        assign y[0] = 2 * x[2];
        assign y[1] = 3 * x[0];
        assign y[2] = 1;
    endmodule
    module test #(parameter WIDTH=8) (
        input [WIDTH-1:0] t
    );
        reg [WIDTH-1:0] hm [0:2];
        reg [WIDTH-1:0] x;
    
        sub #(.WIDTH(WIDTH)) m1(hm, x);
    endmodule

martinwhitaker, 2022-08-09:
Unpacked array ports are a SystemVerilog feature that isn't yet supported. But still, the compiler should output a proper error message rather than failing an assertion.

martinwhitaker, 2022-08-28:
Actually I'm wrong, and there is support for unpacked array ports. The issue here is that in the first case you are trying to connect an unpacked array port to an undefined variable, and in the second case that you are trying to connect an unpacked array port to a packed vector. Both are invalid code, but again the compiler should output a proper error message rather than failing an assertion.

closed 2022-12-21 by steveicarus

larsclausen, 2022-12-22:
With latest develop the example now prints
```
747a.v:15: error: Can not assign non-array identifier `x` to array.
747a.v:15:      : Port 2 (y) of sub is connected to x
1 error(s) during elaboration.
```

PR #820 (2022-12-21): Add error checking for continuous unpacked array assignments 
Currently there is no error checking for continuous assignments to unpacked
arrays. If the lvalue and rvalue net are not compatible undefined behavior
occurs. For some types of incompatibility it will crash during elaboration,
for others it will crash during simulation, and for some it will just work,
even though the assignment is not allowed by the standard.

Implement checking to ensure the two nets are compatible as required by the
standard and report an error otherwise.

Two arrays are considered to be compatible if their elemen [... cut]