steveicarus/iverilog #751: VPI path issue in Windows+Msys

svenka3, 2022-08-06:
When I tried compiling a simple VPI application it works fine on a MAC. When i run the same code in Windows + MSYS2 I got two issues, maybe these are known?

1. iverilog-vpi needdd -mingw=/mingw64 everytime, else I get errors like:

```log
:\Source\iverilog-build\vpi/../../iverilog/vpi/libvpi.c:47: undefined reference to `__imp__assert'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:\tools\iverilog\lib/libvpi.a(libvpi.o): in function `vpi_handle_by_name':
C:\Source\iverilog-build\vpi/../../iverilog/vpi/libvpi.c:55: undefined reference to `__imp__assert'
```

For now I have added -mingw as below:

```sh
	iverilog-vpi -mingw=/mingw64 ../ivlog_vpi_src/az_ivl_uvm_intf_gen.c |& tee comp.log
```
2.  iverilog-vpi produces *.o in current DIR/folder but the *.vpi goes to the DIR where the C source is kept. In my case the transcript looks like:

```log
      | ^~~~
Compiling ../ivlog_vpi_src/az_ivl_uvm_intf_gen.c...
Making ../ivlog_vpi_src/az_ivl_uvm_intf_gen.vpi from  az_ivl_uvm_intf_gen.o...
```
On a MAC it creates the *.vpi in current DIR. 

For now I have copied the *.vpi, but looking for better solution. Also passing the path to *.vpi works as below:

```sh
	vvp -M. -m../ivlog_vpi_src/az_ivl_uvm_intf_gen az_ivl_uvm_intf_gen.vvp
```

But ideally I would like to avoid that - to keep it OS independent as much as possible. Any comments please?

Thanks

martinwhitaker, 2022-08-09:
Are you running in a MSYS2 shell or a MINGW64 shell? The test suite is run in a MINGW64 shell, and the `-mingw` option is not needed there.

`iverilog-vpi` is an executable binary in Windows, but a shell script in Linux and MacOS. So it doesn't surprise me that they don't behave identically.

svenka3, 2022-08-10:
I was using MSYS shell

Should I assume this inconsistency will be as is for a while?

Thanks

martinwhitaker, 2022-08-10:
The instructions on the wiki for building Icarus Verilog using MSYS2 tell you to use a MINGW64 shell because that generates a native Windows application. If you have followed that (or are using a binary obtained from somewhere else), then you need to run iverilog-vpi in the same environment. It should work in both a MINGW64 shell and (if you set up the PATH correctly) in a DOS shell.

I've never tried building Icarus Verilog in a MSYS2 shell, which would generate a MSYS2 application.

I would be wary of altering the Windows version of iverilog-vpi to be consistent with the UNIX version, in case other users are relying on the current behaviour.

martinwhitaker, 2024-01-08:
Works as expected when built as a native Windows application, so closing.

closed 2024-01-08 by martinwhitaker