steveicarus/iverilog #752: vpiHigConn not supported in vpi_handle?

svenka3, 2022-08-07:
I am trying to get HighConn of a port via:
```c
vpiHandle high = vpi_handle(vpiHighConn, sig_h); 
```

I get error on vpiHighConn as undeclared. I looked inside vpi_user.h and it is missing the define for the same. Is this unsupported? Is it easy to add by any chance? 

From standard vpi_user.h I see:
```C
/*********************************** METHODS **********************************/
/**************** methods used to traverse 1 to 1 relationships ***************/

#define vpiCondition          71   /* condition expression */
#define vpiDelay              72   /* net or gate delay */
#define vpiElseStmt           73   /* else statement */
#define vpiForIncStmt         74   /* increment statement in for loop */
#define vpiForInitStmt        75   /* initialization statement in for loop */
#define vpiHighConn         76   /* higher connection to port */
```
In Icarus's version this is missing.

Background - we are trying to enhance debug of OVL in Icarus as part of ABV support in IVL_UVM. So it will help to print the HighConn of a test_expr that fails in an assertion (OVL).

Thanks

svenka3, 2022-08-10:
Thanks, I understand it is not supported as of now. If this is implemented we can do some amount of "connectivity verification " using Icarus - a problem that's becoming prominent these days.

I am curious as a user - as the net value propagation, force etc are working such connectivity information is present inside the database - perhaps it is transformed and hence can't retrieve source code level connectivity ? Is my understanding right?

Cheers
Srini

caryr, 2022-08-10:
Correct, the net is optimized for connectivity and does not currently have access to the source level net information you are looking for.