steveicarus/iverilog #753: sdf_annotate omitted

pi0rock, 2022-08-08:
Hi,

I am trying to run a test on sdf_del_max.v in ivtest/ivltests to test the sdf annotation but i am getting a warning message that the $sdf_annotate is being omitted.

`sdf_del_max.v:21: warning: Omitting $sdf_annotate() since specify blocks are being omitted.`

Below is the command how i run it:
`iverilog sdf_del_max.v`

And the test failed:
Falling took                    0, expected 4
Rising took                    0, expected 3 

Would you be able to advise me on how to run with sdf annotation? Thanks.

caryr, 2022-08-08:
Try adding -gspecify when you invoke iverilog.

pi0rock, 2022-08-09:
Hi @caryr,

Thank you so much!

But i am getting the below warning messages. I am not sure what went wrong.. Could you please point out my mistake? Thanks.

[image]

Command used:
`iverilog -o build/sim -g2012 -gspecify -Tmax -DFUNCTIONAL -DSIM -DGL -DSDF -DUSE_POWER_PINS -I . /tools/scripts/OpenLane/pdks/sky130A/libs.ref/sky130_fd_sc_hd/verilog/primitives.v /tools/scripts/OpenLane/pdks/sky130A/libs.ref/sky130_fd_sc_hd/verilog/sky130_fd_sc_hd.v lab1.gl.v lab1_tb.v`

Below is the RTL
```
lab1 dut (
	`ifdef GL
	// power and ground for GLS with SKY130 cells
		.VPWR(pwr),
		.VGND(gnd),
	`endif

	.clk(clk),
	.a(a),
	.b(b),
	.o(o)
);

initial begin
	`ifdef SDF
		$sdf_annotate("lab1.sdf", dut);
	`endif
end
```

A snapshot of the SDF file:
```
(DELAYFILE
 (SDFVERSION "3.0")
 (DESIGN "lab1")
 (DATE "Mon Aug  1 06:06:31 2022")
 (VENDOR "Parallax")
 (PROGRAM "STA")
 (VERSION "2.3.1")
 (DIVIDER .)
 (VOLTAGE 1.800::1.800)
 (PROCESS "1.000::1.000")
 (TEMPERATURE 25.000::25.000)
 (TIMESCALE 1ns)
 (CELL
  (CELLTYPE "lab1")
  (INSTANCE dut)
  (DELAY
   (ABSOLUTE
    (INTERCONNECT a input1.A (0.013:0.013:0.013) (0.005:0.005:0.005))
    (INTERCONNECT b input2.A (0.020:0.020:0.020) (0.008:0.008:0.008))
    (INTERCONNECT clk clkbuf_0_clk.A (0.035:0.035:0.035) (0.015:0.015:0.015))
    (INTERCONNECT _2_.Y _3_.S (0.000:0.000:0.000) (0.000:0.000:0.000))
    (INTERCONNECT _3_.X _4_.A (0.000:0.000:0.000) (0.000:0.000:0.000))
    (INTERCONNECT _4_.X _7_.D (0.000:0.000:0.000) (0.000:0.000:0.000))
    (INTERCONNECT _5_.Q _3_.A0 (0.000:0.000:0.000) (0.000:0.000:0.000))
    (INTERCONNECT _6_.Q hold1.A (0.000:0.000:0.000) (0.000:0.000:0.000))
    (INTERCONNECT _7_.Q output3.A (0.000:0.000:0.000) (0.000:0.000:0.000))
    (INTERCONNECT _8_.Q _3_.A1 (0.000:0.000:0.000) (0.000:0.000:0.000))
    (INTERCONNECT clkbuf_0_clk.X clkbuf_1_0__f_clk.A (0.001:0.001:0.001) (0.000:0.000:0.000))
    (INTERCONNECT clkbuf_0_clk.X clkbuf_1_1__f_clk.A (0.000:0.000:0.000) (0.000:0.000:0.000))
    (INTERCONNECT clkbuf_1_0__f_clk.X _8_.CLK (0.000:0.000:0.000) (0.000:0.000:0.000))
    (INTERCONNECT clkbuf_1_0__f_clk.X _5_.GATE_N (0.000:0.000:0.000) (0.000:0.000:0.000))
    (INTERCONNECT clkbuf_1_1__f_clk.X _7_.CLK (0.000:0.000:0.000) (0.000:0.000:0.000))
    (INTERCONNECT clkbuf_1_1__f_clk.X _6_.GATE (0.000:0.000:0.000) (0.000:0.000:0.000))
    (INTERCONNECT hold1.X _5_.D (0.000:0.000:0.000) (0.000:0.000:0.000))
    (INTERCONNECT input1.X _8_.D (0.000:0.000:0.000) (0.000:0.000:0.000))
    (INTERCONNECT input1.X _2_.B (0.000:0.000:0.000) (0.000:0.000:0.000))
    (INTERCONNECT input2.X _6_.D (0.000:0.000:0.000) (0.000:0.000:0.000))
    (INTERCONNECT input2.X _2_.A (0.000:0.000:0.000) (0.000:0.000:0.000))
    (INTERCONNECT output3.X o (0.001:0.001:0.001) (0.001:0.001:0.001))
   )
  )
 )
 (CELL
  (CELLTYPE "sky130_fd_sc_hd__nand2_1")
  (INSTANCE _2_)
  (DELAY
   (ABSOLUTE
    (IOPATH A Y (0.057:0.057:0.057) (0.048:0.048:0.048))
    (IOPATH B Y (0.075:0.075:0.075) (0.070:0.070:0.070))
   )
  )
 )
 (CELL
  (CELLTYPE "sky130_fd_sc_hd__mux2_1")
  (INSTANCE _3_)
  (DELAY
   (ABSOLUTE
    (IOPATH A0 X (0.117:0.117:0.117) (0.274:0.274:0.274))
    (IOPATH A1 X (0.118:0.118:0.118) (0.285:0.285:0.285))
    (IOPATH S X (0.191:0.192:0.193) (0.315:0.316:0.317))
    (IOPATH S X (0.130:0.130:0.131) (0.306:0.306:0.307))
   )
  )
 )

```

caryr, 2022-08-09:
See #746.

pi0rock, 2022-08-10:
Noted. Thank you @caryr

closed 2022-08-10 by pi0rock