steveicarus/iverilog #755: Assertion failed VPI

ToTamire, 2022-08-11:
My simulation in cocotb sometimes fail to run with the following error:

> Assertion failed: vpi_mode_flag != VPI_MODE_NONE, file ../../iverilog/vvp/vpi_priv.cc, line 1286

Error is totally random. Sometimes simulation run correctly 10 times in a row and other times simulation fails 10 times in a row.

Environment:
- Icarus Verilog Parser/Elaborator version 12.0 (devel) (s20150603-1539-g2693dd32b)
- cocotb 1.6.2 x64
- conda 4.12.0
- Python 3.9.12

martinwhitaker, 2022-08-11:
Without a simple test case that allows us to reproduce the problem, there's not a lot we can do to help you. We might be able to deduce something if you provided a gdb backtrace from when the assertion failed.

steveicarus, 2023-02-26:
Lack of feedback from the reporter, so closing this issue.,

closed 2023-02-26 by steveicarus