steveicarus/iverilog #759: VPI - vpiFile returns wrong information

svenka3, 2022-08-22:
Hello,
  I have a simple VPI based Lint code at: https://github.com/svenka3/ivl_uvm/blob/main/ivl_uvm_vpi/ivl_vlog_lint/check_name.c

When I run on Windows/Msys2 I found that vpiFile returns module name than file name. Interestingly (and luckily) vpiLineNo works fine. 

Code snippet:

```c
/* Get the file where module is decalred */
  file_name = vpi_get_str(vpiFile,argHandle);

```

Is this a known limitation? 

Thanks

caryr, 2022-08-22:
That sounds like a bug where the buffer is not getting cleared correctly. Can you provide a complete code snippet?

caryr, 2022-08-22:
Actually this is  bug in your code. From the standard:
```
The VPI routine vpi_get_str() shall return string property values. The string shall be placed in a temporary
buffer that shall be used by every call to this routine. If the string is to be used after a subsequent
call, the string should be copied to another location.
```
My guess is resolving this will get things working as expected.

Basically add a `strdup()` and then `free()` on error or when done with the value.

svenka3, 2022-08-23:
Thanks a lot Cray for ponting it out. I fixed it and works like charm:

```log
IVL_UVM: module: dut instance: wrong_named_dut FullName: M.wrong_named_dut
ERROR :instance Declaration doesnt start with u*_ 'wrong_named_dut'
           File : test.v, line no : 15
```

closed 2022-08-23 by svenka3