steveicarus/iverilog #760: Issue with nmos simulation of a flip flop

ead0601, 2022-08-25:
While reverse engineering an IC, a flop flop nmos representation was extracted from the layout. The icarus simulation provides an incorrect response to its functionality. This testing was re-simulated with other tools that provide the correct response. The same testbench and verilog code was used with each simulator. (note one of the simulators used was an industry EDA tool.)  

FILE: https://github.com/ead0601/mioc-verilog/blob/main/mioc-flop/src/mioc_flop_nmos.v

TESTBENCH: https://github.com/ead0601/mioc-verilog/tree/main/mioc-flop

In the testbench execute either "make run" or "make waves" (gtkwave)

This is a custom flip flop for an IC from the 80s for the ADAM computer. It has four inputs , of which none are a clock, async set and reset, "sync"(delayed) reset, and delayed inverted reset.

Thanks
Ed

ead0601, 2022-08-25:
Out of the four inputs, the in2 "negedge reset" is a reset that occurs on the negedge  of the in2 transition. The functionality of this input "in2" is incorrect.  Run make waves and look at the transitions.

ead0601, 2022-08-25:
The issue is with mioc_flop_mos (implemented with nmos),  I created a mioc_flop_rtl, that provide the functionality needed for the latch. That version simulates correctly. Thanks.

martinwhitaker, 2022-08-27:
After a quick look, I suspect there's a race condition in your circuit. Replacing `nmos` with `nmos #1` gives the behaviour you expect, as does moving the `LEVEL 1` section to after `LEVEL 2` (without adding any delays).

If you still think there's a bug in Icarus, please try to create a simpler example that demonstrates it.

ead0601, 2022-08-31:
Yes, adding the delay fixed the issue. Thanks