steveicarus/iverilog #776: Inferred latch in `always_comb` instead of `x` generation

mkorbel1, 2022-09-27:
In the below example, when the `if` condition is not true, the driven output `bytes` is still valid as the previous valid value instead of `x`.  I have seen in other examples that rather than a latch being inferred, an `x` is driven on the un-driven signals.

I'm running on "Version 11.0 (stable)".

This was found during debug of [ROHD](https://github.com/intel/rohd) issue https://github.com/intel/rohd/issues/158 since ROHD uses iverilog to confirm that generated SystemVerilog is logically equivalent to the ROHD simulator.  The example below was modified slightly from the original ROHD-generated code to simplify for this example.  It may be possible to reduce it further.

```SystemVerilog
module ExampleModule(
input logic [20:0] codepoint,
output logic [31:0] bytes
);
logic [20:0] original___codepoint_rshift__const_6_multiply__swizzled_add_const_224;
logic [1:0] count;
//  combinational
always_comb begin
  if((codepoint == 21'h2020)) begin
      count = 2'h2;
      bytes = ({24'h0,(original___codepoint_rshift__const_6_multiply__swizzled_add_const_224 & 8'hff)});
      count = (count - 2'h2);
  end 

end

assign original___codepoint_rshift__const_6_multiply__swizzled_add_const_224 = (codepoint >> (5'h6 * ({3'h0,count}))) + 21'he0;  // add
endmodule : ExampleModule
module tb;
logic [20:0] codepoint;
logic [31:0] bytes;
ExampleModule dut(.codepoint(codepoint), .bytes(bytes));
initial begin
  $dumpfile("bug_waves.vcd");
#1
codepoint = 8224;
#2
if(bytes !== 32'he2) $error($sformatf("Expected bytes=32'he2, but found bytes=0x%x with inputs {codepoint: 8224}", bytes));
#8
codepoint = 32;
#2
if(bytes !== 32'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx) $error($sformatf("Expected bytes=32'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx, but found bytes=0x%x with inputs {codepoint: 32}", bytes));
#8
codepoint = 8224;
#2
if(bytes !== 32'he2) $error($sformatf("Expected bytes=32'he2, but found bytes=0x%x with inputs {codepoint: 8224}", bytes));
#8
codepoint = 32;
#2
if(bytes !== 32'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx) $error($sformatf("Expected bytes=32'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx, but found bytes=0x%x with inputs {codepoint: 32}", bytes));

$finish;
end
endmodule
```

martinwhitaker, 2022-09-28:
I see nothing in the IEEE standard that says this behaviour is not correct. The standard does say

> Software tools should perform additional checks to warn if the behavior within an always_comb procedure
does not represent combinational logic, such as if latched behavior can be inferred.

so yes, we should output a warning, but that is all.

mkorbel1, 2022-09-28:
I found another example where there were X's but it looks like it was before any valid signal was driven on the conditionally undriven signals, so I guess the behavior is consistent with what I saw before and I just hadn't looked closely enough.

Thanks for clarifying expectations, you're right it sounds like a warning is all that's required.