steveicarus/iverilog #782: C-style comment after `include causes malformed include error message

yuri-panchul, 2022-10-08:
C-style comment after `include causes malformed include error message:

`include "b.v" /* c  */

a.v:1: error: malformed `include directive. Did you quote the file name?

yuri-panchul, 2022-10-08:
Same issue with `timescale:

`timescale 1ns / 10ps /* set time scale                                                                          */

testbench.v:21: Invalid `timescale directive (extra garbage after precision).

steveicarus, 2022-12-16:
Martin fixed this, but never closed the issue. Closing it now.

martinwhitaker, 2022-12-17:
I left this open because I only fixed it for `include. Fixing it for `timescale (and for the other directives handled by the compiler rather than by the preprocessor) is a bigger job, and I haven't had time to tackle it.

martinwhitaker, 2022-12-21:
Both single and multi-line comments are now supported after an include directive but not interspersed with it. The directives supported by the compiler itself (resetall, timescale, default_nettype, [no]unconnected_drive, [end]celldefine) are now properly free-form, so are not restricted to one line and can have comments freely interspersed.

The directives that aren't currently supported (pragma and the various optional and non-standard directives) behave as before - they are only recognised if there is nothing but white space before them on the line, and then the whole line is ignored. That is probably good enough for common usage (it handles single line comments), so I don't propose to do more work on that unless there's a real need.

closed 2022-12-21 by martinwhitaker