steveicarus/iverilog #788: Widely accepted non-ANSI style task argument declarations are now treated as errors

martinwhitaker, 2022-11-09:
A simple example:
````
module test();

task print_hex;

  input n;

  reg [7:0] n;

  begin
    $display("%h", n);
  end  
endtask

initial begin
  print_hex(66);
end
   
endmodule
````
This code compiles without errors or warnings on all other simulators I've tried. But since PR #767, it is rejected by Icarus Verilog with the error message:
````
test.v:7: error: Scalar port ``n'' has a vectored net declaration [7:0].
````
Unlike for module ports, the IEEE standard doesn't specify what is/isn't allowed.

I found this when compiling the Micron DDR3 DRAM model, so there is real code out there that uses this form of declaration.

larsclausen, 2022-11-09:
There is `-gno-io-range-error`, which will turn this into a warning. I supposed we can turn that on by default.

martinwhitaker, 2022-11-14:
Given that IEEE 1364-2005 section 12.3.7 includes this example:
````
module driver (net_r);
  output net_r;
  real r;
  wire [64:1] net_r = $realtobits (r);
endmodule
````
I do think it is wrong to treat it as an error by default, or even to output a warning. After all, the only reason for continuing to support non-ANSI style port declarations is to maintain backwards compatibility. This was common practice, and all the big name simulators still support it without a murmur.

steveicarus, 2022-11-15:
We could turn it around, and have it be a warning that we turn on with
-Wall. Or not. I'm OK either way.

martinwhitaker, 2022-12-18:
OK, I've resolved this by making the warning dependent on the anachronisms warning category being enabled, so there is at least a way to suppress it.

I found an anomaly whilst testing this. The anachronisms warning category is enabled by default. That's not what I would expect from reading the iverilog man page, which states it is included by `-Wall`.

closed 2022-12-18 by martinwhitaker