steveicarus/iverilog #791: How to debug iverilog

WXY9902, 2022-11-17:
Hello, I would like to know how to debug iverilog. Because I didn't find it on the official docs. Can you please give me some help.

caryr, 2022-11-17:
We will need significantly more information to help you. What issue are you experiencing, do you have a short code example that demonstrates the issue? Also have you enabled warnings? They can provide useful information.

WXY9902, 2022-11-18:
Sorry, I don't have the corresponding code. I'm just wondering if iverilog has debug mode, can you give me some instructions accordingly.

steveicarus, 2022-12-11:
This issue is far too vague. Closing. If there is something more specific needed, please create a more detailed issue.

closed 2022-12-11 by steveicarus