steveicarus/iverilog #793: Compilation terminated by assertion error

EDA-Testing, 2022-11-28:
Hi When I used Iverilog, an assertion error occurred and the compilation was terminated. I can't figure out where the specific problem is, the error message is as follows
`**ivl: net_link.cc:130: Nexus* Link::find_nexus_() const: Assertion `next_' failed.**`
Wondering if you could give me some info on this.

larsclausen, 2022-11-28:
Hi, thanks for reporting the issue. Can you provide an small example code snippet that is able to reproduce to the issue. This will help to better understand what is causing the issue.

cemery123, 2022-11-28:
Hi My code snippet may be a bit long, I'm uploading it now and I hope it won't have too much extra impact on you

cemery123, 2022-11-28:
This is my design file：
```
module top  (y, clk, wire3, wire2, wire1, wire0);
  output wire [(32'h89):(32'h0)] y;
  input wire [(1'h0):(1'h0)] clk;
  input wire signed [(4'hb):(1'h0)] wire3;
  input wire [(4'hb):(1'h0)] wire2;
  input wire signed [(4'h8):(1'h0)] wire1;
  input wire signed [(4'h9):(1'h0)] wire0;
  wire signed [(4'hb):(1'h0)] wire22;
  wire signed [(3'h7):(1'h0)] wire21;
  reg signed [(2'h3):(1'h0)] reg20 = (1'h0);
  reg [(4'h9):(1'h0)] reg19 = (1'h0);
  reg [(4'ha):(1'h0)] reg18 = (1'h0);
  reg [(4'h8):(1'h0)] reg17 = (1'h0);
  reg signed [(3'h7):(1'h0)] reg16 = (1'h0);
  reg [(2'h3):(1'h0)] reg15 = (1'h0);
  reg [(4'h8):(1'h0)] reg14 = (1'h0);
  wire signed [(3'h5):(1'h0)] wire13;
  reg [(4'h8):(1'h0)] reg12 = (1'h0);
  reg [(3'h6):(1'h0)] reg11 = (1'h0);
  reg [(4'h9):(1'h0)] reg10 = (1'h0);
  reg [(3'h7):(1'h0)] reg9 = (1'h0);
  reg signed [(3'h5):(1'h0)] reg8 = (1'h0);
  wire [(3'h5):(1'h0)] wire7;
  wire [(3'h7):(1'h0)] wire6;
  wire signed [(4'h9):(1'h0)] wire5;
  wire signed [(4'h9):(1'h0)] wire4;
  assign #5 y = {wire22,
                 wire21,
                 reg20,
                 reg19,
                 reg18,
                 reg17,
                 reg16,
                 reg15,
                 reg14,
                 wire13,
                 reg12,
                 reg11,
                 reg10,
                 reg9,
                 reg8,
                 wire7,
                 wire6,
                 wire5,
                 wire4,
                 (1'h0)};
  assign #5 wire4 = ((+wire1) ? $unsigned(wire2) : $unsigned(wire1));
  assign #5 wire5 = $signed(wire0);
  assign #5 wire6 = (~^(8'h9d));
  assign #5 wire7 = $unsigned((+(wire6 ^~ $signed(wire6))));
  always
    @(posedge clk) begin
      reg8 <= wire6;
    end
  always
    @(posedge clk) begin
      reg9 <= wire0[(3'h5):(3'h5)];
      if ($signed(({(~wire6)} - (((8'h9d) ? wire0 : wire1) ?
          $unsigned((8'ha3)) : wire7))))
        begin
          reg10 <= $signed((-(8'hac)));
        end
      else
        begin
          reg10 <= (wire6[(3'h4):(1'h1)] || $signed($signed(wire0[(3'h6):(2'h3)])));
        end
    end
  always
[... 58 lines cut ...]
      #10 {wire3, wire2, wire1, wire0} = (255'h4ed9bce0ccb085c51d16ae7e13b8d944d3df6adccd088fdc5aa8d4bbbb6fc77e);
      #10 {wire3, wire2, wire1, wire0} = (254'h2ee0782628a87f76fadcd8885a1477e51d56908202186e0db63a4b87338867df);
      #10 {wire3, wire2, wire1, wire0} = (256'hc3c1c03530854dcb91cdb1ab4aab46fe1eeea111ba42420f81626001ad6de9a7);
      #10 {wire3, wire2, wire1, wire0} = (256'hde73348c70efa3afc6a02b1a9d9dbe5c46fc10a5f4a49657261b4f3876175acf);
      #10 {wire3, wire2, wire1, wire0} = (255'h73f17dab00d009b96ed24dea44b1769888abfef275d57de84a50436135123707);
      #10 {wire3, wire2, wire1, wire0} = (254'h304c7a5949e9e0b123e02b9534887615f818b3f87df40cda8d53a2c8cb290ffc);
      #10 {wire3, wire2, wire1, wire0} = (250'h3f679ea45eaba7289c3fcd355c1de35133b804c70dd1eb8b782d76317e24634);
      #10 {wire3, wire2, wire1, wire0} = (256'he1a6fcdcf3da5bd18eb2278181c5f0fddddac29660c712c39ae44f616b265d7e);
      #10 {wire3, wire2, wire1, wire0} = (256'h90b96f1c4b4513a70f4de630efd1c498ccb8e570d3a4a5106ebf5de5cbb0191a);
      #10 {wire3, wire2, wire1, wire0} = (256'h8a65e0111131243004a2bc9a48cadcd3620ba3700c193eac17781563e2504124);
      #10 {wire3, wire2, wire1, wire0} = (256'hf8cccfb00a3b6fc5ce39d8b282b28a1ae99b8ea1f4481cae7a564ce0531968aa);
      #10 {wire3, wire2, wire1, wire0} = (256'hd630e31cc43bea2760cc81c69f00a1aa799aae2bce90430dcec20f1856da7bf1);
      #10 {wire3, wire2, wire1, wire0} = (254'h23dcc5d17b059a4844f293c5e9aa8a26af11347ea18a37179f13a9af56eac664);
      #10 {wire3, wire2, wire1, wire0} = (255'h454b4c493d917ddb43675d025262ed02de3c668ae5050fd5c0eeee1b54a57834);
      #10 {wire3, wire2, wire1, wire0} = (256'hc85cdb9d865ecfed4febbb9f1a033a67f520e22021710dcd5425e5f99935ab85);
      #10 {wire3, wire2, wire1, wire0} = (256'hbaac74b417eeb733d2e2eb3d22708af2765fc4605554b7c219db13dcd4ba3a01);
      #10 {wire3, wire2, wire1, wire0} = (256'h8c0926432811f1de288f3a7a7065f0136744cdbf4564abd3ea4888302185f8a7);
      #10 $finish;
    end
  always
    #5 clk = (~clk);
  always
    @(posedge clk) $strobe ("%b", y);
endmodule
```

larsclausen, 2022-11-28:
Thanks for providing the sample. I'd reduced it to a minimum that still triggers the issue. It happens when there is an assignment with a delay and also using either $signed or $unsigned on the right hand side of the assignment.
```
module top ();
    wire [9:0] a, b;
    assign #1 a = $signed(b);
endmodule
```

cemery123, 2022-11-29:
Hi, thank you for the extra reduction that saved us a lot of time in identifying the problem.

But in similar operations, we also encountered the problem of adding $sign() after Assignment, and in this case, it did not bring about the termination of Iverilog compilation. The specific documents are as follows:
```
module top
#( parameter param13 = ({(!((8'had) >= (8'had)))} ? (~^({(8'h9e)} >> ((8'ha7) << (8'h9d)))) : (((~^(8'ha2)) ? (^~(8'ha3)) : ((8'h9c) ? (8'had) : (8'ha2))) ? ({(8'hac)} - (^~(8'ha4))) : {((8'ha6) == (8'had))})) )
(y, clk, wire3, wire2, wire1, wire0);
  output wire [(32'h4a):(32'h0)] y;
  input wire [(1'h0):(1'h0)] clk;
  input wire [(4'h9):(1'h0)] wire3;
  input wire [(4'h8):(1'h0)] wire2;
  input wire [(4'hb):(1'h0)] wire1;
  input wire [(2'h3):(1'h0)] wire0;
  wire signed [(4'h8):(1'h0)] wire12;
  reg signed [(4'ha):(1'h0)] reg11 = (1'h0);
  reg [(4'ha):(1'h0)] reg10 = (1'h0);
  reg signed [(3'h4):(1'h0)] reg9 = (1'h0);
  reg [(3'h4):(1'h0)] reg8 = (1'h0);
  wire [(3'h7):(1'h0)] wire7;
  reg signed [(4'hb):(1'h0)] reg6 = (1'h0);
  reg [(4'h9):(1'h0)] reg5 = (1'h0);
  reg signed [(4'ha):(1'h0)] reg4 = (1'h0);
  assign #5 y = {wire12,
                 reg11,
                 reg10,
                 reg9,
                 reg8,
                 wire7,
                 reg6,
                 reg5,
                 reg4,
                 (1'h0)};
  always
    @(posedge clk) begin
      reg4 <= wire1[(3'h6):(2'h2)];
      if ({({$unsigned(wire3)} ? (wire2 == reg4) : $signed((wire1 >>> wire2)))})
        begin
          reg5 <= wire1[(4'h8):(2'h3)];
          reg6 <= $signed(wire3);
        end
      else
        begin
          reg5 <= reg4;
          reg6 <= $signed(({{reg4}} ?
              reg4[(2'h3):(2'h3)] : reg6[(4'h8):(2'h2)]));
        end
    end
  assign #5 wire7 = ($unsigned((((8'ha2) && reg5) ?
                         (reg6 <= wire3) : $signed((8'ha4)))) ?
                     ((^reg6[(3'h7):(3'h5)]) >= $signed((~^wire3))) : (&reg6));
  always
    @(posedge clk) begin
      reg8 <= {$unsigned(wire3)};
      reg9 <= ($signed($unsigned({wire3})) - wire2[(1'h0):(1'h0)]);
      if (wire3)
        begin
          reg10 <= (({$signed((8'haa))} <<< $unsigned((wire2 >= (8'ha1)))) != ($signed({reg8}) ?
              $signed(reg9) : $signed(reg9[(2'h2):(2'h2)])));
          reg11 <= reg10[(2'h3):(1'h1)];
        end
      else
        begin
          reg10 <= reg6[(4'h8):(3'h4)];
        end
    end
  assign #5 wire12 = (({$unsigned(wire0)} ?
                          reg10[(3'h4):(3'h4)] : $unsigned({wire3})) ?
                      $signed(($unsigned(reg10) >> (reg11 != wire3))) : {({wire2} ?
                              $unsigned((8'hae)) : $signed(wire0))});
endmodule
[... 24 lines cut ...]
      #10 {wire3, wire2, wire1, wire0} = (254'h2436b24a97c4c249253e7023ccd21bddd413845e89c7413cf55a0779258a5127);
      #10 {wire3, wire2, wire1, wire0} = (255'h557e91d142184788602bfc44df6784ca7c05421273e5ed1f00312304d93513cf);
      #10 {wire3, wire2, wire1, wire0} = (256'hc4bfa4ba143b423452b72f32636b676cb187c9d028ca24589854ebb11f611bd2);
      #10 {wire3, wire2, wire1, wire0} = (256'hb184cb752dee9c9273192421d1f743733a349119daf232d19360e2ed394d4cd5);
      #10 {wire3, wire2, wire1, wire0} = (255'h7a526fd7b3ef69ba118aaa3aa437a5eafdde38f79fde40cc7bbd3b1afbb88f9d);
      #10 {wire3, wire2, wire1, wire0} = (256'ha70749cc97f154183f22a75b65069efecd130eb9c2d9c9e0d5cc9c9ed115dbf0);
      #10 {wire3, wire2, wire1, wire0} = (256'h8aeb6c53d751fdf74b02f97188060fd6057855b60e54d5d59e66253a42cffdf1);
      #10 {wire3, wire2, wire1, wire0} = (256'hae9429f58dbc6b90233fc8493f40776ed1068ed4f6735be88573756754f409ff);
      #10 {wire3, wire2, wire1, wire0} = (256'h8b97c789a62f878f9498c683ba29ff1024a866898f11d374aa31b0940d7eb217);
      #10 {wire3, wire2, wire1, wire0} = (256'hb2cd82a3d76ed793fe96042ba227df73bc5966a1b5b4b15cd527df23ad33e584);
      #10 {wire3, wire2, wire1, wire0} = (253'h11144ed5e0fb5078be94c7e3821b895c696bf687f5a15d2b185c1703b127fdd2);
      #10 {wire3, wire2, wire1, wire0} = (256'hc273c2a9e8578b7fa725104809b563fc10cfaac4849eb60ec841181c5cec67b7);
      #10 {wire3, wire2, wire1, wire0} = (256'h84c00c07767812b6902c7acc4f4abc0baebe50677c12817f874fdd9d920f9b6f);
      #10 {wire3, wire2, wire1, wire0} = (253'h147488e9c82fb32ac16977bcdf54c89658437bce945d84b1581e634b1f4f9939);
      #10 $finish;
    end
  always
    #5 clk = (~clk);
  always
    @(posedge clk) $strobe ("%b", y);
endmodule
```
We don't know if this is caused by the inconsistent behavior of iverilog's compiler in similar situations. In addition, if the mandatory symbol conversion function $sign() cannot be added to the right side of the assignment equation, should the iverilog compiler prompt a related warning instead of crashing directly during the compilation process in our case?

Sorry to bother you. Also, if you have a kit for reducing code, can you share some with us?

larsclausen, 2022-11-30:
The issue is clearly a bug and needs to be fixed. The problem only happens if the right hand side is a sign cast on a undriven net.

E.g. this works
```SystemVerilog
wire a = 1'b1;
wire b = $signed(a);
```
while
```SystemVerilog
wire a;
wire b = $signed(a);
```
does not.

I created a PR that fixes the issue, but I'm really not sure if it is the right approach. See #795.

For reducing the sample I removed half the lines and tested whether the issue still occurs if it does remove the next half and so on. If it doesn't pick the other half that was just removed and continue with only that.

steveicarus, 2022-12-12:
I merged PR#795, so I suspect that this issue is now resolved. Can you-all confirm that I can close this issue?

steveicarus, 2022-12-16:
I ran the original example, and it didn't crash the compiler. I'm calling this "fixed." Re-open if you disagree.

closed 2022-12-16 by steveicarus