steveicarus/iverilog #794: PAL example fails with ERROR: Unable to read config file

robertrau, 2022-11-29:
Using the example:
https://github.com/steveicarus/iverilog/blob/master/examples/pal_reg.v
and the suggested command line in the comments
iverilog -tpal -ppart=generic-22v10-plcc -opal_reg.jed pal_reg.v
I get the error
ERROR: Unable to read config file: /usr/local/Cellar/icarus-verilog/11.0/lib/ivl/pal.conf
error:  failed to load.
: dlopen(/usr/local/Cellar/icarus-verilog/11.0/lib/ivl/, 1): no suitable image found.  Did find:
	/usr/local/Cellar/icarus-verilog/11.0/lib/ivl/: not a file
	/usr/local/Cellar/icarus-verilog/11.0/lib/ivl: not a file
error: Code generator failure: -2

I could not find the file pal.conf anywhere.

Thank you,

Bob Rau

martinwhitaker, 2022-12-03:
tgt-pal is not built by default. As it has not been maintained for many years, even if you do try to build it, it may not work.

robertrau, 2022-12-03:
Understand. Thank you.

closed 2022-12-03 by robertrau