steveicarus/iverilog #822: VPI Interface Issue

TomO700, 2022-12-20:
Hi @steveicarus 

I am trying to upgrade a working icarus/cocotb test bench to newer versions and it seems that test data is generated but not injected to the DUT. I don't have any errors but I do get the following warnings. Was suspecting it was related to icarus since the warning comes from VPI.

```
~/.local/lib/python3.9/site-packages/cocotb_bus/drivers/__init__.py:92: DeprecationWarning: This method is now private.
  self._thread = cocotb.scheduler.add(self._send_thread())
     0.00ns WARNING  gpi                                VPI: Not able to map type vpiNamedBegin(33) to object.
     0.00ns WARNING  gpi                                Unable to create $ivl_for_loop0 via any registered implementation
~/.local/lib/python3.9/site-packages/cocotb_bus/monitors/__init__.py:67: DeprecationWarning: This method is now private.
  self._thread = cocotb.scheduler.add(self._monitor_recv())
```

The details of the system

_5.13.0-48-generic #54~20.04.1-Ubuntu SMP x86_64 x86_64 x86_64 GNU/Linux
cocotb 1.7.2
python 3.9.5
**Icarus Verilog version 11.0**_

Thank you

steveicarus, 2022-12-21:
I don't recognize those error messages. Can you narrow down what you think the vpi is doing wrong?

TomO700, 2022-12-22:
Hi @steveicarus 
These are warnings and data is not getting injected on the SV side even though it is successfully generated on the python side. I have checked this via the waveform as well. How do you suggest I narrow this down?

TomO700, 2022-12-22:
@steveicarus 
Can this cause any issues?

`apt list -a iverilog` returns
`iverilog/focal,now 10.3-1build1 amd64 [installed]`

`iverilog -`V returns
`Icarus Verilog version 11.0 (stable) (v11_0-130-gaf2b95276-dirty)`

drmuelr, 2024-10-09:
I'm also seeing this problem with what appears to be the same environment as @TomO700. I'll follow up with a minimal example once I'm past some imminent deadlines.

caryr, 2024-12-29:
Can you provide example code that reproduces the issue so we can use it to track the issue down? The message "VPI: Not able to map type vpiNamedBegin(33) to object." is from `cocotb` and specifically src/cocotb/share/lib/vpi/VpiImpl.cpp. If we cannot get some example code I will close this issue.

My guess is the `cocotb` side cannot find a way to get at the Icarus object and that is why things are not working. We may not need a complete working example, but as a minimum we would need some Verilog code to show the context of the access and then what `cocotb` is specifically trying to connect to.

caryr, 2025-07-15:
No test code provided so closing.

closed 2025-07-15 by caryr