steveicarus/iverilog #840: Two assertions firing when dealing with generate with arrays of structs

wmlye, 2022-12-28:
Again running in WSL2, and I did `git pull` about 10 minutes ago, so this is an issue reporting against 13.0  I'm attempting to code an efficient binary tree based search similar to how I've coded Wallace trees in the past.  I pass in an array [N-1:0] of the structure to search, then I allocate an array [2*N-1] of the same structure., then I fill the first N positions of the local structure with the input elements (leaving the N-1 elements at the end empty) then do a N-1 comparisons, each comparison comparing adjacent elements, chosing one, and copying it to the next empty position.

Anyways, I tried using generate to construct this (as I have in the past using Verilog) and I found that I could trigger two different assertions in `elab_expr.cc`.  I used generate instead of always_comb to try to get around some of the other assertions I've been having "fun" with.

The first assertion is as follows:

```
wlye@AMR-BU-L02782:~/foton/tests/issue-xxx$ cat struct_array_assert_test1.sv
typedef struct packed {logic x;} s;

module struct_array_assert_test1
(
  input  s [1:0]   a
);

s [1:0] b;

generate
  genvar i;
  for(i=0;i<=2;i++)
    assign b[i] = a[i];
endgenerate

endmodule
wlye@AMR-BU-L02782:~/foton/tests/issue-xxx$ iverilog -g2012 struct_array_assert_test1.sv
struct_array_assert_test1.sv:13: assert: elab_net.cc:467: failed assertion rcl
Aborted
wlye@AMR-BU-L02782:~/foton/tests/issue-xxx$
```

For the second, if I change the definition of b from being 2 elements (i.e. N) to being 3 elements (i.e. 2*N-1) I get this:

```
wlye@AMR-BU-L02782:~/foton/tests/issue-xxx$ cat struct_array_assert_test2.sv
typedef struct packed {logic x;} s;

module struct_array_assert_test1
(
  input  s [1:0]   a
);

s [2:0] b;

generate
  genvar i;
  for(i=0;i<=2;i++)
    assign b[i] = a[i];
endgenerate

endmodule
wlye@AMR-BU-L02782:~/foton/tests/issue-xxx$ iverilog -g2012 struct_array_assert_test2.sv
struct_array_assert_test2.sv:13: assert: elab_expr.cc:6223: failed assertion rc
Aborted
wlye@AMR-BU-L02782:~/foton/tests/issue-xxx$
```
The equivalents using always_comb instead of generate don't cause these same errors.

wmlye, 2022-12-28:
As an aside, I suspect that if I keep messing around in the intersection of arrays, structures, and generate I may well find more similar issues; would you prefer I raise them as separate issues here or would you prefer I add them to an omnibus issue?

larsclausen, 2022-12-29:
I think this is a case of constant out-of-bounds indices when indexing into a dimension that is not the last dimension. E.g. a slightly simpler testcase to trigger the same issue is.
```SystemVerilog
module test;
  logic [0:0][0:0] x;
  logic a;
  assign a = x[1];
  assign x[1] = a;
endmodule
```

Instead of triggering an assert this should just issue a warning and replace the out-of-bounds value with an `x`.

wmlye, 2022-12-29:
Hmm...   In the first example both a and b have the same dimensions.  In the second example they have different dimensions, but it's b that is bigger (not a) such that b[2] is left unassigned.  Replacing the generate/endgenerate with always_comb, changing the genvar to int, and removing the assign results in .sv files that compile correctly without errors/warnings/assertions.

wmlye, 2022-12-29:
Ah, now I see it; I'm using `<=2` rather than `<2`, so yes indeed it's going off the end.  With that in mind I've got four testcases to better show the effect and a `<3` to better show the run-off-the-end behavior:
```
wlye@AMR-BU-L02782:~/foton/tests/issue-836$ cat test1.sv
module test1;
  logic [1:0][1:0] a;
  logic [1:0][1:0] b;
  generate
    for(genvar i=0;i<3;i++)
      assign b[i] = a[i];
  endgenerate
endmodule
wlye@AMR-BU-L02782:~/foton/tests/issue-836$ cat test2.sv
module test2;
  logic [1:0][1:0] a;
  logic [2:0][1:0] b;
  generate
    for(genvar i=0;i<3;i++)
      assign b[i] = a[i];
  endgenerate
endmodule
wlye@AMR-BU-L02782:~/foton/tests/issue-836$ cat test3.sv
module test3;
  logic [1:0][1:0] a;
  logic [1:0][1:0] b;
  always_comb
    for(int i=0;i<3;i++)
      b[i] = a[i];
endmodule
wlye@AMR-BU-L02782:~/foton/tests/issue-836$ cat test4.sv
module test4;
  logic [1:0][1:0] a;
  logic [2:0][1:0] b;
  always_comb
    for(int i=0;i<3;i++)
      b[i] = a[i];
endmodule
wlye@AMR-BU-L02782:~/foton/tests/issue-836$ iverilog -g2012 test1.sv
test1.sv:6: assert: elab_net.cc:467: failed assertion rcl
Aborted
wlye@AMR-BU-L02782:~/foton/tests/issue-836$ iverilog -g2012 test2.sv
test2.sv:6: assert: elab_expr.cc:6229: failed assertion rc
Aborted
wlye@AMR-BU-L02782:~/foton/tests/issue-836$ iverilog -g2012 test3.sv
wlye@AMR-BU-L02782:~/foton/tests/issue-836$ vvp a.out
wlye@AMR-BU-L02782:~/foton/tests/issue-836$ iverilog -g2012 test4.sv
wlye@AMR-BU-L02782:~/foton/tests/issue-836$ vvp a.out
wlye@AMR-BU-L02782:~/foton/tests/issue-836$ 
```
All four of these testcases are incorrect and should probably fire error or a warning.  `test1`/`test2` are using `generate`/`endgenrate` and the `for()` loop is evaluated by `ivl`. while `test3`/`test4` are use `always_comb` so the `for()` loop is evaluated by `vvp`.  `test1`/`test3` run off the end of the boundary on both LHS and RHS, while `test3`/`test4` just run off the end of the RHS.  The LHS/RHS difference explains the two different assertions thrown by `ivl` in `test1`/`test2` so they are related but not the same.

Now I'm wondering about `test3`/`test4`.  In both files, the code is accessing beyond the array in the `for()` loop; what is the "correct" thing to do for this?  Somehow I was expecting that `vvp` would warn me that I'm attempting to access beyond the bounds of the array, but it's not doing that.  However when I modify `test3` to make it a 1D array I notice that this doesn't result in `vvp` giving me any warnings either.  I suspect that there is a historical reason for this (maybe matching what other simulators do) that I'm not aware of.
```
wlye@AMR-BU-L02782:~/foton/tests/issue-836$ cat test3a.sv
module test3;
  logic [1:0] a;
  logic [1:0] b;
  always_comb
    for(int i=0;i<3;i++)
      b[i] = a[i];
endmodule
wlye@AMR-BU-L02782:~/foton/tests/issue-836$ iverilog -g2012 test3a.sv
wlye@AMR-BU-L02782:~/foton/tests/issue-836$ vvp a.out
wlye@AMR-BU-L02782:~/foton/tests/issue-836$ 
```
If this is to match other simulators, would it make sense to add a (optional) command line option to `vvp` to strictly enforce boundaries like this?  Errors like this are so easy to create and if this option were available I would always enable it.

I've separately verified that if I replace `<3` with `<2` in all four cases, everything is clean.

larsclausen, 2022-12-30:
The difference between test1/2 and test3/4 is that for test1/2 and with the genvar the index is a constant and known at elaboration time. For test3/4 it is a runtime variable and will only be evaluated at runtime. Currently there is no warning when a out-of-bounds happens at runtime. For reads an `x` will be returned, writes will be discarded. This is the behavior required by the standard. The standard says that implementations may also issue a warning in addition. And this would a nice feature to have that can be enabled with a switch, but currently there is no support for it.

For the asserts that you are seeing at runtime have a look at 7d7aa0604c49d24b35081e5db571e8e9e86b9a75. The same change needs to be done at the places where the asserts are triggered.

wmlye, 2022-12-30:
Gotcha.  I probably should have been more specific about compile-time (i.e. "in `ivl`") vs. run-time (i.e. "in `vvp`") :-)

I may just make those modifications and make a submission this weekend, i.e. treat it as a "training wheels" exercise before attacking #835.

wmlye, 2022-12-30:
OK, I've created a pull request addressing this issue and the issue from #823, replacing the assertions with error messages.

On the issue of producing warnings when attempting to access outside an array, I've read through the code and by my understanding most of the magic would happen in `vvp/vthread.cc/of_STORE_VEC4()` and possibly in its %store/vec4a sibling `of_STORE_VEC4A()`.  Writing beyond the bounds is dealt with by truncating the bounds with `vvp/vthread.cc/resize_rval_vec()`, and reading beyond the bounds (and Z-padding) is handled by `vvp/vvp_net_sig.cc/vvp_fun_signal4_sa::recv_vec4_pv()`.  The writing-beyond-the-bounds problem is easy to detect because all the necessary numbers are already extracted from the data structure for me, but detecting the reading-beyond-the-bounds problem is a bit trickier; I need to read through the data structure a bit more before I can figure it out.

martinwhitaker, 2022-12-31:
We also have the problem that the run-time doesn't properly handle out-of-bounds access for multi-dimensional arrays. This is an old bug - https://sourceforge.net/p/iverilog/bugs/953/

wmlye, 2022-12-31:
> We also have the problem that the run-time doesn't properly handle out-of-bounds access for multi-dimensional arrays. This is an old bug - https://sourceforge.net/p/iverilog/bugs/953/

Ho boy, that looks like a challenge, and I suspect it's baked into how the VM is implemented; it looks like an array

```
logic [L-1:0][M-1:0][N-1:0] array;
```
is just allocated as a variable in the vvp-tgt output:
```
address .var "a" L*M*N-1 0;
```
then indexing into the array is done using a set of `%load/vec4`/`%padd/s`/`%muli`/`%add`/`%addi'/'%part' to get the source, then another set of index calculations and an `%ix/vec4/s 4;` to set destination index, followed by a `%store/vec4 address, 4, width;` to write to the array.  In other words, the `vvp` VM has no knowledge of how the array is laid out.  That makes sense from an efficiency perspective, but if we were to try to add bounds checking to it, we would have to add multiple `%cmp`/`%cmpi` opcodes to the emitted code which would slow things down.  Maybe we could add an extra opcode or two to allow more-efficient bounds checking?

However that sounds a bit like putting a band-aid on it.  I'm looking at some of the emitted code from some of my testcases.  Admittedly these involve very short loops so the loop overhead is large relative to the body of the loop, but I'm seeing quite a few redundancies in the emitted code that come from using a stack-based VM that might be alleviated if it were based on a load/store-based VM.  I'm wondering if it is better to band-aid the VM or would it make sense to create a new VM, leaving the `vvp` VM as-is.  Of course, at that point the question of VM design comes up, and I'm sure someone would suggest LLVM; I'm no-where expert enough on VM's or on the intricacies of LLVM and how that would interact with VPI to have an informed opinion on that.  And then, of course, there's the question about whether or not anyone wants to become the VM lead; VM's are historically quick and fun to write but absolutely horrible to debug and maintain...

wmlye, 2022-12-31:
Thinking about it a bit more...

functors' inputs and outputs are all statically determined at compile time so `ivl` can provide meaningful feedback, i.e. it can do bounds checking and can give meaningful error messages with useful names.  However, for the rest of it that has to be interpreted by the `vvp` VM, the fact that everything (numbers, structures, arrays, or hierarchical combinations of all three) is treated as canonical arrays of bits means that the `vvp` code doesn't know and doesn't how its laid out, so its ability to give meaningful feedback is severely limited.

Most of the time the expectation is that we would get that feedback by monitoring the outputs in either a text dumpfile (manually or with a script) or manually in a graphical viewer, but for the array overflow errors described in that bug report it would be much better/easier if that could be caught by the VM, especially if it can give meaningful warnings.  The warnings message modifications to `of_STORE_VEC4()` and now I'm realizing to the opcode handler for `%part` that I'm contemplating would only give messages that something wrapped around so much that it exceeded the total length of the variable, and can only meaningfully say "you accessed outside the range of variable X", and can only give that variable's root name, i.e. nothing about what its internal structure is.

We might be able to give that information by emitting decoration pseudo-ops giving what amounts to an AST for how the variable is laid out, and the hypothetical bounds checking opcodes I describe in the last comment could access that information, but this is sounding like a band-aid on top of a band-aid.

caryr, 2022-12-31:
When I thought about this a long time ago my idea was to have a new opcode to calculate the normalized unsigned index along with all the appropriate bounds checking. It could set a flag to indicate the out of bounds condition that would then be used to return X or skip the assignment.

wmlye, 2022-12-31:
@caryr Having a single opcode to do the index calculation for all dimensions simultaneously would be cool, but the VM opcode parsing in `compile.cc` can only handle three values passed in at a time and we would rapidly run out of room.  Maybe have one opcode per dimension, something like this:
```
logic [N-1:0][M-1:0] a;
int i,j;
logic x;
assign x = a[i][j];
```
Could be emitted like this:
```
a     .var "a" N*M*L-1 0;
i      .var "i" 31 0;
j      .var "j" 31 0;
x     .var "x" 0 0;

       %load/vec4 a;
       %pushi 0;                       start index at 0
       %load/vec4 i;
       %bounds N-1 0 M;    add i*M to index, apply bounds checking
       %load/vec4 j;
       %bounds M-1 0 1;       add j to index, apply bounds checking
       %part/s 1;
       %store/vec4 x 0 1;
```

`%bounds a,b,c;` would pop a value off the stack, make sure it's between a and b, and if so multiply it by c and add it to the next value at the head of the stack.  If it's out of bounds, it sets a flag on the value on the stack indicating it's out-of-bounds.  The `%part` can examine the flag and do X substitution and issue a warning.  Using `%bounds` isn't all that much different from what we've got today; it's replacing several `%mul`/`%add`/`%pad` and also adding bounds checking without needing to add several `%cmp%`'s as well.  It also scales easily to more than two dimensions; a 3-dimensional array would have three `%bounds` opcodes.

Doing the opposite isn't all that much different
```
assign a[i][j] = x;

       %load/vec4 x;
       %pushi 0;                       start index at 0
       %load/vec4 i;
       %bounds N-1 0 M;    add i*L*M to index, apply bounds checking
       %load/vec4 j;
       %bounds M-1 0 1;       add j to index, apply bounds checking
       %ix/vec4/s 4;
       %store a, 4, 1;
```
the `%bounds` opcodes are exactly the same, the only difference is the `%part` is replaced by `%ix` and which variable is associated with the first `%load` and the last `%store`.

%bounds` is probably a pretty bad name (it's doing much more than just bounds checking, but that's the first thing I thought of) and I may well be slightly mangling some semantics of various bits and pieces of the VM opcodes but hopefully you get the idea.

If any of the dimensions are known at compile time, we can do bounds checking then and replace `%bounds` with combinations of `%muli` and/or `%addi`.

wmlye, 2023-01-01:
OK, getting back to the second part of my original problem (i.e. not the larger question of bounds checking of multiple sub-indexes) I've confirmed that I can detect over-runs on reading from a bit vector in `vvp/vthread.cc/of_PART_base()`, and I've confirmed that I can detect over-runs on writing to a bit vector in `vvp/vthread.cc/of_STORE_VEC4()` and these correctly detect the errors in my original testcases. However I'm stuck on how to make useful warning messages.

The perhaps-easier-to-track-down issue is in `of_STORE_VEC4()`. I'm given the pointer to the destination variable to be written to, but I'm trying to figure out how to work backwards from that to the name. `vvp/words.cc/compile_variable()` is given the label and the name during compilation. The label is stored in the `vvp_net_t` object and is used for lookup purposes in all further compilation stages, while the name stored in a `vpiHandle` object that has a link to the net.  If we have a name, we can look it up in the vpiHandle and then easily find the net, however if we have a net we can't easily get the vpiHandle and therefore get the name.  I'm guessing we would have to start with the top-level list of `vpiHandle`s and then find the one that contains the link to the net, but I don't have any idea where to start doing that. Is there an already-existing function to do this?

The potentially much harder issue is in `of_PART_base()`. In this case, it's working with the source value that has already been stored on the stack through a `%load/vec4` opcode at the start of the index calculation.  However my read of the code (which isn't guaranteed to be correct) is that when the `%load/vec4` is done it pushes the net value onto the stack (not the net itself) so all I've got at this point is a value which has been disconnected from its net.  If I could go from a value back to a net, I could then do the same `vvp_net_t` => `vpiHandle` lookup as in `of_STORE_VEC4()`, however if we were instead to store the net on the stack (not the value) we would need to implement a copy-on-write mechanism so that if we change the value on the stack (there's a lot of manipulation of the value on the top of stack throughout the code) we don't change the value associated with the net.

Of course, once that's figured out I would need to add infrastructure for enabling/disabling warnings (would you recommend I port the `ivl` warning infrastructure over to `vvp` or do something else?) and then probably add a mechanism for muting a warning after it's already fired on a given net.  And of course, this may all be moot if we start adding opcodes to the VM...

caryr, 2023-01-01:
I didn't read all of this yet, but my idea was you would pass the array variable since it already knows all the bounds and then pass how many bounds have been pushed onto the stack. General array selects make it more complicated so all that needs to be considered, though we could make one opcode to handle single indexes with possible full arrays at the end and then one that allows a range for each index. At the moment we don't handle array selects so that implementation can be pushed off, but planning how it all should work should be done now so things work well once we have a need for it.

wmlye, 2023-01-01:
Hmmm...  I've just realized that I've been exclusively talking about packed arrays.  Do unpacked arrays (especially multi-dimensional unpacked arrays combined with packed arrays like the `bit [1:5][1:6] v4 [1:7[1:8];` example from Section 7.4.5 of 1800-2012) throw any wrenches into the gears here?

Doing some tests, it looks like iverilog creates variables that are packed arrays using `.array` with the dimension based on the first packed dimension.  When accessing data from the packed array, it first calculatess the canonical index in the unpacked dimensions then directly accesses the element using `%load/vec4a`.  From there it treats it just as if it were a packed array, calculating the canonical index for the packed dimensions followed by `%part`.  Have I got that right?

Is there anything preventing `ivl`/`vvp` from treating packed and unpacked dimensions the same, using the same access mechanism for both?  I.e. if I were to define two variables `logic [1023:0][31:0] mem1;` and `logic [31:0] mem2 [1023:0];` is there any practical reason to handle them differently? 

Also, I've realized that there is potentially a significant performance penalty when accessing packed arrays.   As they are currently implemented; packed arrays are accessed through `%load/vec4`...`%part`, which means that value stored in the net is copied bit-by-bit by `vvp/vvp_net_sig.cc/vvp_fun_signal4_sa::recv_vec4()` onto the stack, and is then eventually pulled apart `%part` once we know what the index is.  If someone has a particularly large packed array (like `mem1` above) and we are only accessing a very small portion of it (especially inside `for()`/`foreach()`/`generate` loops) this means the entire array value is continually being copied and then discarded.  If `%load/vec4` were to instead push the net onto the stack rather than the net's value, then I think we would be safe if everything that uses values from the stack can work interchangably with net values or bare values, and anything that modifies the top element in the stack (i.e. anything that does a `peek_vec4()`) does a copy-on-write, replacing nets with values.  This way we never write the full-width element to the stack, just the portion that has been extracted by `%part`.  The semantics are the same, just there there will be much less copying of values from nets to the stack.  There may even be a performance improvement when accessing regular variables too.

My discussion with myself (I've never claimed to be sane) has taken this issue far away from its original intent.  I'm going to raise a separate issue discussing bounds checking of array accesses in vvp, and will bring up the copy-on-write there.

martinwhitaker, 2024-01-08:
Appears to have been fixed by PR #845, so closing.

closed 2024-01-08 by martinwhitaker