steveicarus/iverilog #846: Support multidimensional parameters

catkira, 2022-12-31:
Is it correct that iverilog does not support multidimensional parameters like this?
```verilog
module mod1
#(
    parameter [31:0] PARAM1 [0:2] = '{32'd1, 32'd2, 32'd3}
)
(
    input clk_i
);
```
I am using version 11.0 (stable)

martinwhitaker, 2022-12-31:
Yes. Parameters having an unpacked array type is a SystemVerilog feature that isn't supported yet.

catkira, 2022-12-31:
ok. How hard do You think would it be to implement? I would give it a try if its not too enormous.

catkira, 2022-12-31:
@larsclausen told me he is alraedy on it, so I just leave it open until we hear great news 8-)

smunaut, 2023-09-18:
Any progress on this ? Just got hit by it ...

Kreijstal, 2024-06-16:
Just got hit by this, so here to chime in :)

Kreijstal, 2024-07-13:
maybe related #654

cousteaulecommandant, 2024-11-12:
> Yes. Parameters having an unpacked array type is a SystemVerilog feature that isn't supported yet.

What about parameters with a **packed** array type?  Should those work?  Because so far I'm out of luck.

I posted a description of the issue in #1180 but perhaps I should've created a separate issue for that bug (or simply posted it here).