steveicarus/iverilog #848: Support for run-time bounds checking of multi-dimensional arrays

wmlye, 2023-01-01:
Continuation/Discussion from #840.

In my original problem statement I brought up the fact that we don't have run-time warnings generated from `vvp` when accessing data outside the bounds of multi-dimensional packed arrays.

I've made some proof-of-concept hacks to `vvp` to confirm that yes I can detect when we are reading data outside an array (loaded onto the stack with '`%load/vec4`) with `%part` or writing data outside an array with `%store/vec4`.  However creating meaningful error messages for either is difficult because by the time `%part` has been executed, the information about which net the data was read from has been lost, and when `%store/vec4` runs the VM has information about the address of the `vvp_net_t', but not the name which has been stored in a `vpiHandle` structure.  `vpiHandle` has a pointer to the `vvp_net_t`, but the `vvp_net_t` doesn't have a pointer to the `vpiHandle`.  I will raise this as a separate issue.

@caryr brought up https://sourceforge.net/p/iverilog/bugs/953/, which brings up the possibility of wrapping around inside multi-dimensional array indexes without going beyond the array boundaries, i.e. if we have a packed array `logic [3:0][1:0] a;` we have a canonical array size of 8 bits, and the bits at `a[i][j]` are accessed through canonical index `k=2*i+j`, i.e. `a[3][0]`, `a[2][2]`, `a[1][4]`, and `a[0][6]` all reference the same location.

The current implementation of the VM has no information about how variables are laid out; when we access an element from the array, the value of the entire packed array is first loaded onto the stack using `%load/vec4`, the canonical index is calculated and left as the top element on the stack, then the index is pulled from the stack and the array value (the top element of the stack) is modified to pick only the bits we want using `%pick`.  When storing a value into an array element, first the array element is pushed onto the stack, the canonical index is calculated and stored in index register 4, then the value is written to the variable.

In order to do the bounds checking without VM changes, `tgt-vvp` would need to emit multiple `%cmp` opcodes on every canonical index calculation, which would definitely slow things down.  If people are open to making VM changes, I suggested that we could create a new opcode that can be used to do canonical index bounds checking, multiplication, and addition all in one, allowing each dimension to be checked and calculated in a single opcode, so if there are three dimensions the opcode would be emitted three times.  @caryr said that he had an idea where everything could be done all dimensions at once with a single opcode.  I think either way could be made to work, but we need to emit information to give `vvp` information on bounds limits, either through decoration pseudo-ops to go with `.array` or with parameters to the new opcodes.

Beyond that is the question of unpacked arrays, especially multi-dimensional unpacked arrays.  Unpacked arrays are defined with `.array` with a width corresponding to the product of the packed dimensions and a length corresponding to the total size, unlike packed arrays that are defined with `.var` with a length set just by the total size.  By my understanding, `vvp` reads from unpacked arrays by first calculating the canonical index of the unpacked dimensions and storing it index register 4, loading the selected element onto the stack using `%load/vec4a` (taking advantage of the width defined with `.array`), then calculating the canonical index and pulling out the desired bits using `%pack` just as is done for packed arrays.   Storing to packed arrays seems to be handled by calculating the canonical unpacked index and storing it in index register 4,  calculating the canonical packed index and storing it in index register 5, then storing to it with `%store/vec4a` using both index registers 4 and 5.

larsclausen, 2023-01-01:
> If people are open to making VM changes, I suggested that we could create a new opcode that can be used to do canonical index bounds checking, multiplication, and addition all in one, allowing each [...]
100% agree. If the bounds check is false set flag 4 to 1'bx. If the array access instructions sees that flag 4 is x it will ignore the read/write. We could make the index instruction so that it modifies that index register instead of an on stack value.

The only problem I see is that the instruction needs two constants. What to multiply the value with and what to use for the out-of-bounds check.
```SystemVerilog
int a[2][3][4];
a[i][j][k] = 10;
```
would translate to
```
%ix/getv 4, &label_to_k
%load/vec4 &label_to_j # push j to stack
%ix/array 4, 4, 3 # does index[4] += j * 4; if (stack_top < 0 || stack_top >= 3) flag[4] = 1'bx;
%load/vec4 &label_to_i # push i to stack
%ix/array 4, 12, 2 # does index[4] += stack_top * 12; if (stack_top < 0 || stack_top >= 2) flag[4] = 1'bx;
```

For this to work, we'd need special NetExpr to represent the array index build-up.

wmlye, 2023-01-01:
Rather than storing it into index 4, I was thinking of building it up on the stack which wouldn't need a special NetExpr; check out my (admittedly ugly) example in #840.

larsclausen, 2023-01-01:
> Rather than storing it into index 4, I was thinking of building it up on the stack which wouldn't need a special NetExpr; check out my (admittedly ugly) example in #840.

I think you still need the special NetExpr to encode the array bounds. Something like ```NetArrayIndex(NetExpr *base, NetExpr *index, unsigned long mul, unsigned long size);``` Which does `base + index * mul` and also the bounds check. If you have a special NetExpr for array indices you can directly operate on the index register, since that is a bit faster than doing the math on a vector where you have to do conversions from vector to integer.

wmlye, 2023-01-02:
Ah, that makes sense.  Either way, I'm back to my day job on Tuesday so my bandwidth to hack around will be somewhat reduced, but my interest is definitely piqued.  However, before taking on something like this I think I still need more time to become familiar with the code base and the class structure.  To that end I'm thinking that spending some time tweaking the codebase comments to be more Doxygen-friendly might be a good idea.  I ran Doxygen on the entire structure and it gave me some breadcrumbs, but also gave me a lot of spaghetti with no context, so I also spent a lot of time with good old `grep`.  As @caryr points out we need to avoid problems with VPI, so I'm thinking that `vvp` would be a good place to start that task.